barbeau/gpstest #58: Galileo support

tomte, 2016-04-28:
Hi!
I bought a phone that supposedly has a galileo positioning module built in. Unfortunately I have no idea how to verify that. How can I check with this app which positioning systems I can access?

barbeau, 2016-04-28:
@tomte Nice!  Just install the [app on Google Play](https://play.google.com/store/apps/details?id=com.android.gpstest), and then take a screenshot and post it here (just drag/drop into issue).

Based on the satellite ID values shown in the app, we should hopefully be able to figure out if some previously unknown IDs show up.  Unknown IDs will still show up with an American flag next to them, as that's the current default behavior in the app.  If we can identify the ID range for Galileo satellites, I can update app to show an EU flag next to Galileo satellites.

More [here](http://gis.stackexchange.com/questions/65412/is-there-an-industry-standard-official-mapping-of-glonass-satellites-to-prn-va) on why I need the ID range of Galileo satellites.

tomte, 2016-04-28:
doesn't look good...

[image]

barbeau, 2016-04-28:
Hmmmm, yes, 1-32 should be GPS IIRC, and 70-88 is GLONASS.  So nothing unidentified yet.  

What city/country are you located in?  It could be possible that your device supports Galileo but there aren't any Galileo satellites currently visible to your device.  Also, can you share your device make/model (e.g., Samsung Galaxy S6)?

tomte, 2016-04-28:
The location is sweden and it is a snopow m9. I guess I have glonass inbuilt instead of galileo...

barbeau, 2016-04-28:
That may be the case.  From some quick Googling it seems the only GNSS listed in the device specs is GPS.

tomte, 2016-04-28:
Anyway - thank you for the help. If I get my hands on something with galileo I let you know!

barbeau, 2016-04-28:
Sounds good, thanks!  I'll leave this issue open so anyone else with Galileo can chime in as well :).

bluezed, 2016-09-19:
Hi there,
just wondering if there has been any progress with this?
I've just bought a new BQ Aquaris X5 Plus which is supposed to be able to receive Galileo data but with your App I can only see GPS, GLONASS and Baidou.
Let me know if there's anything I can do to help with testing or similar.
Thanks!

barbeau, 2016-09-19:
@bluezed Awesome!  Yes - could you please post a screenshot of the main Status screen here, after your device has acquired a fix?

The Galileo satellites will likely currently show up as US NAVSTAR, as that is the default if the ID for the satellite isn't recognized.  So, any satellites outside of the known US NAVSTAR ID range showing up as NAVSTAR should be Galileo.

Here's the IF block that controls what a satellite gets labeled as based on it's ID:
https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/java/com/android/gpstest/util/GpsTestUtil.java#L27

bluezed, 2016-09-19:
Here's a quick screenshot. 
[image]

Doesn't look like there are any "NAVSTAR" IDs. Could that be because the Galileo PRNs appear to be pre-fixed with the letter "E" and in your code you're expecting int values?

Is there anywhere to view the "raw" data in you app? Maybe that could be useful...

barbeau, 2016-09-19:
The sats with PRNs from 1-32 with the American flags are the USA's NAVSTAR GPS system.  If the Android platform was providing information to the app about Galileo satellites, they should show up with an American flag (for now) with a PRN outside of 1-32.  I would then update the app to show an EU flag for those PRNs to add "Galileo support" - but you'd be able to see the Galileo raw data there before I make any changes.

Here is a good wiki page that discusses the IDs for the known PRN ranges for other GNSS:
https://github.com/mvglasow/satstat/wiki/NMEA-IDs#galileo

>Could that be because the Galileo PRNs appear to be pre-fixed with the letter "E" and in your code you're expecting int values?

The Android platform itself uses integer values in the original GPS.Listener - https://developer.android.com/reference/android/location/GpsSatellite.html#getPrn().

However, looking at the Android docs, Google did recently add a new API where you can extract the GNSS type explicitly - https://developer.android.com/reference/android/location/GnssStatus.html

I'll look at updating the app to take advantage of this, although it would only be available on Android N 7.0 and higher.

What version of Android is on your device?

bluezed, 2016-09-20:
Thanks for the explanation. Unfortunately I've only got Android 6 but the rumour is that Android 7 will come to this phone at some point as well.

I've now raised a ticket with the phone manufacturer to clarify why I cannot see Galileo satellites so we'll see what they say. 

From what I can make out there's currently no Android app that shows Android satellites for the on-board sensor, only some that can use external devices so maybe it is currently a platform limitation...?

barbeau, 2016-09-20:
>From what I can make out there's currently no Android app that shows Galileo satellites using the on-board sensor, only some that can use external devices so maybe it is currently a platform limitation...?

Based on my experience, I think you're correct.  Android 6 and lower was designed primarily for providing NAVSTAR GPS (USA) satellite info, which have numeric PRNs (satellite IDs).  They crammed GLONASS, Beidou, and QZSS into this same model by mapping satellites to numbers outside of the NAVSTAR PRN range.  It seems like Galileo IDs are alphanumeric, which means they wouldn't fit into this same model.  It seems like instead of trying to continue this trend of mapping other GNSS satellites to numeric IDs, Google decided to deprecate the old API and move to a new one that explicitly labels the satellites with the GNSS constellation, and then can provide the IDs that are unique to that GNSS but not globally unique.  This is a better solution as it better represents the real world, but unfortunately because it requires Android 7.0 it's going to take a while for devices to catch up.

barbeau, 2016-09-20:
I've added an answer to my Galileo question on GIS StackExchange here, which details the new satellite IDs for Android 7.0 and higher:
http://gis.stackexchange.com/a/211229/12198

mbavaro, 2016-09-22:
Hi all I am also following this thread closely now. I agree that better multi-GNSS support is given in Android Nougat, but the chipset is already capable of tracking Galileo with special PRN identifiers so once the base-band firmware is enabled it should show up on the satellite list (or the NMEA senstences anyway). I think Aquaris 5X does not have the Galileo-enabled baseband yet and it will be provided with an OTA later this year. But I noticed now that Galaxy Note 7 and Tp-Link Neffos X1 also claim Galileo support. Would be good if somebody could try them for us here.

bluezed, 2016-09-22:
Quick update about what I got back from BQ support...

Basically the current firmware on the device is "deaf" to the Galileo signal and hence the data does not show up even though the hardware is capable of receiving it already. Once Galileo is actually operational they'll release an OTA update to enable it on the phone.

So it is as suspected and my guess is that the update will then be to Android 7 at the same time. Wait and see ;)

mbavaro, 2016-10-07:
Would it be possible to have an EU flag for PRNs 301-330?

[image]

More importantly, how to log all the raw NMEA then?

barbeau, 2016-10-07:
@mbavaro Yes!  Thanks for sharing!  Can you provide any more information on the device this screenshot came from, or how you know the range is 301-330?  It would be great to have an official reference for this.

A NMEA logger has been on my TODO list for a while (https://github.com/barbeau/gpstest/issues/30), although that will probably come with a future release, as part of a UI overhaul.  The current tabbed UI doesn't offer a great place to put this.

mbavaro, 2016-10-07:
I cannot say too much.. it is indeed a preview of the Aquaris X5+. The choice of PRN 300+ is arbitrary as far as I know. If not logging NMEA, how could one record all the observations taken by your app - or some similar app if you are aware of one? I have searched Google play but most seem bloatware.. I just need a text log file to parse.

barbeau, 2016-10-07:
Sure, understandable.

If you're able to install [Android Studio](https://developer.android.com/studio/index.html), I could probably pull together a quick build that would just dump the NMEA output to the system log.  To see this, you'd connect your device to your machine via USB cable, open Android Studio, and look at Android Monitor - from here you could copy the system output out of Android Monitor into a text editor (see screenshot below).  It wouldn't be a pretty UI (actually no UI), but it would at least let you see the NMEA output.

Would this work?

[image]

mbavaro, 2016-10-10:
Hello Sean, it's probably too early to jump into doing more work on this (also in perspective of the soon to come changes brought by Nougat). Let me first see if I can perform my duties without wasting too much of your time. I will anyway keep you up-to-date with my findings. Again a big thank you for applying the Galileo delta.

barbeau, 2016-10-10:
FYI for those following this thread - I'm in the process of setting up a beta channel via Google Play to push the new changes to support Galileo. I'll try to publish this new version tonight - I'll also post more instructions then about how to sign up for the beta.

closed 2016-10-10 by barbeau

mbavaro, 2016-10-10:
Wow.. thanks!

barbeau, 2016-10-11:
Beta testing group is now set up,  and you can sign up to get beta releases via Google Play by following these instructions:
https://github.com/barbeau/gpstest/blob/master/BETA_TESTING.md

However,  I ran into a few issues publishing a new release that I need to sort out. I'll post again when I actually push out the new beta release that contains Galileo and Android N support.

barbeau, 2016-10-12:
Alright, I've published a new version (v2.1.4) that includes Galileo support and dumping NMEA output to system logs as a beta (visible in Android Studio via Android Monitor).  Instructions to sign up for beta are at https://github.com/barbeau/gpstest/blob/master/BETA_TESTING.md.  I'll work on adding documentation soon for installing Android Studio and seeing NMEA output in Android Monitor and will post here when that's updated.  Feedback welcome!

barbeau, 2016-11-08:
Just realized that I never updated this thread - Galileo support is now in production release on Google Play, and docs for setting up logging via Android Studio are here:
https://github.com/barbeau/gpstest/blob/master/LOGGING.md

nicoladinisio, 2017-04-13:
Dear Sean,

I continue on this old post on Galileo Support. I have a BQ Aquaris X5 plus with

GPS Test 2.1.10
Android 7.1.1 (patch level March 1st, 2017)

As of now I can see GPS, GLONASS and BEIDOU satellites, but no Galileo is shown, even though at the moment 11 Galileo satellites are reported as operational by the GSA

https://www.gsc-europa.eu/system-status/Constellation-Information

barbeau, 2017-04-13:
Thanks for the update! Have you tried waiting outside with a clear view of
the sky for at least five minutes? You might try different times of the day
as well.

Sean

nicoladinisio, 2017-04-13:
I have tried again right now.
7 Minutes, I got plenty of GPS, GLONASS and three BEIDOU.
On the sky view I see only circles (GPS), squares GLONASS and pentagons
(BEIDOU).
With 11 spacecrafts in operations right now, I would have expected to see
at least one Galileo passing over my location :)
I suspect that something is wrong with the config of my phone...

bluezed, 2017-04-13:
Hi guys,

I'm in Nauders, Austria today and I see 3 Galileo satellites. Attached a
screenshot.

Cheers

bluezed, 2017-04-14:
Hi guys, 
I'm in Nauders, Austria today and I see 3 Galileo satellites. 
Attached a screenshot...

[image] 

EDIT: Now there are even 5 ;) 

[image]

nicoladinisio, 2017-04-14:
What phone do you have?
I have tried with two BQ Aquaris X5 Plus (with Android 7.1.1 and GPS Test
2.1.10) and both fail to show Galileo spacecrafts.
Tried again today, more than 15 minutes.

bluezed, 2017-04-14:
Same phone as you with same versions and all.

I'm near Meran, Italy now and can see 3 Galileo satellites right now.

[image]

nicoladinisio, 2017-04-14:
What are the numbers in the column after the flags. Some spacecraft are
missing a number in that column, especially all the Galileo ones

barbeau, 2017-04-15:
Column after the flags is carrier-to-noise-density ratio (C/N0) on Android 7.0 and higher. This determines if the receiver can lock onto and decode information from the satellite radio waves (larger numbers are better). 

If this is blank, it's very likely that the receiver can't actually see a signal for that satellite. 

The columns to the right of C/N0 are Elevation and Azimuth of the satellite, or its approximate position in the sky - this comes from the Galileo almanac information. In a standalone GNSS receiver (i.e., no assistance information) the receiver would need to decode the almanac information from the GNSS signal itself - so, no signal, no almanac. However, networked devices can speed up time to first fix by downloading the almanac info directly from a server. This is what Android devices do (and you can actually download the almanac data for Galileo yourself at https://www.gsc-europa.eu/system-status/almanac data. Looks like it was last updated about 3 days ago). 

So, it appears the device was able to grab the Galileo almanac data from somewhere, which is why there is an "A" in the right-most column, and why you can see the Elevation and Azimuth info.  However, it seems like it can't actually see/decode any live Galileo signals, hence why C/N0 is blank.

If you're not seeing any Galileo almanac info on your device, try making sure you have a good Internet connection, and then power off and on your device. When it's back up, again confirm you're connected to the Internet, and start up GPSTest again.  I'm not exactly sure of the mechanism used to fetch assistance data on all devices/networks today (I assume it's SUPL, and I'd also assume that for official Android devices they are using a Google SUPL server), but this could differ per cell provider as well if your device is locked to a particular provider.

In GPSTest you can try tapping on the 3 dots in the upper right corner and choosing "Inject XTRA Data", but I'm guessing that won't have any effect on assistance data (if not, it will say "Platform doesn't support injecting XTRA data") . I don't believe that XTRA data is used anymore for most modern devices - all Android 7.0 devices I've seen so far don't support it.

nicoladinisio, 2017-04-15:
Thank for the thorough explanation Sean!

I guess that what my two devices are missing is the Almanac information.
Yesterday and today I notice the appearance of one satellite at the bottom
with US-flag (which I understand is the default one for unknown satellites)
and this was carrying no flag, in particular the A, for the Almanac

[image: Inline images 3]

This might be a Galileo satellite.
I still do not understand why my two Aquaris X5 Plus do not have Almanac
information, one of the two is new, the other though has been in service
for 6 months now.

barbeau, 2017-04-18:
@nicoladinisio The satellite at the bottom is likely SBAS, or an unknown.  As long as you have Android 7.0 or higher (and your device OEM has implemented the API correctly) all Galileo sats should be labeled with EU flag.

As to missing Almanac and lack of Galileo signals - I'd strongly suggest you contact your device OEM and as what you should be able to see given the Android/firmware version on your device (and report back here :) ).  They are really the only one that will be able to answer the question and give specifics.

Nippey, 2017-08-02:
Screenshot from Galaxy S8+ (SM-G955F) / Stock Android 7.0
[image]

Came here from [gis.stackexchange](https://gis.stackexchange.com/questions/174225/android-devices-and-apps-with-support-for-galileo-positioning/174313?noredirect=1#comment394289_174313)

barbeau, 2017-08-03:
Thanks @Nippey!

kanchudeep, 2017-08-14:
Screenshot taken on OnePlus 5 (Android 7.1.1/OxygenOS 4.5.8):
[image]

Note: Reception is not continuous... Comes and goes...

BurhanBudak, 2021-09-23:
Is this still an issue for older phones. I suspect that some 820/821 phones support Galiloeo unofficially because they dont want to pay up for license that the SoC manufacturer offers.

barbeau, 2021-09-23:
@BurhanBudak Yes, it's still an issue for older phones. Many phones never got an update to support Galileo even if the chipset specs say it's supported. You can see what devices from around the world are reporting in the crowdsourced GPSTest Database - see https://barbeau.medium.com/crowdsourcing-gnss-capabilities-of-android-devices-d4228645cf25 for details.

BurhanBudak, 2021-09-23:
> @BurhanBudak Yes, it's still an issue for older phones. Many phones never got an update to support Galileo even if the chipset specs say it's supported. You can see what devices from around the [...]

I submitted a idea but I will repeat my theory. How can I be sure there is no false/positive connection. Could it be that there is unofficial support for Galileo.

I did a experiment. As Lineage is more true to AOSP, I thought I would get CF flags when I used GPSTest. Nothing changed between OOS and LOS. But then I tried your oldest app possible to download from Github and the newest. Reason for why I tested was to check if I got any unknown or Galileo sats as NAVSTAR flagged. Read in you older bug reports that it would happen. Behold I saw one satellite as unknown with the number 38. Also one falsely flagged NAVSTAR between 100-128, I dont remember nr. I believe I will get more data when the ROM I use switches to Android 11. The missing API really makes it hard to find out.

barbeau, 2021-09-23:
>Could it be that there is unofficial support for Galileo.

It's possible but doubtful, although it would be nearly impossible to prove without having access to the chipset source code that computes the position solution to know what's being computed. There have been a few bugs where OEMs have shown a satellite in the Status list but haven't set the "U" flag to indicate that it's being used. But I don't know of a case where a satellite didn't show up in the Status list but was used to compute a position solution by the chipset.

>Edit: By disabling GNSS.API I got more different data and could distinguish GNSS. 

At this point I wouldn't trust anything coming from the older GPS APIs. The GPS APIs are now officially deprecated and will apparently stop working when apps target Android 12 in their configuration. I expect OEMs are paying minimum attention to these APIs now, and we have a hard enough time getting quality data out of the officially reported APIs.

>As Lineage is more true to AOSP

Just be careful interpreting results when flashing custom ROMs, things are liable to break if there isn't an exact match between the driver and what Android is expecting.

BurhanBudak, 2021-09-24:
> It's possible but doubtful, although it would be nearly impossible to prove without having access to the chipset source code that computes the position solution to know what's being computed. There [...]

I have researched on porting and looking at  budget chipset for my current phone and I found nada solutions. But I maybe found one very **dangerous** solution. As the modem is the controller for the connectivity stack, modifying NV values would "maybe" fix this if the support is in the phone. If a sat signal is like a 4g band then _maybe_ but its easier to just buy a new phone now. When I see no use for my older phone I will try it out as I have "acquired" QC software. It cant be hardware related block. We have seen OEM use external GNSS chips since forever because QC is expensive.

> At this point I wouldn't trust anything coming from the older GPS APIs. The GPS APIs are now officially deprecated and will apparently stop working when apps target Android 12 in their [...]

Would you want the screen shots for the unknown number 38 sat? Because I have never seen a blank sat before in the app.

> > As Lineage is more true to AOSP
> 
> Just be careful interpreting results when flashing custom ROMs, things are liable to break if there isn't an exact match between the driver and what Android is expecting.

Using a unofficial ROM gives more options of API that where not present before but API are there but not used? I there anything the community can do as aftermarket support?

BurhanBudak, 2021-09-24:
> It's possible but doubtful, although it would be nearly impossible to prove without having access to the chipset source code that computes the position solution to know what's being computed. There [...]

Here is a dumpsys location dump on pastebin. I deleted location info and sensitive data. I do see constalations but the phone cant translate the data.
https://pastebin.com/ejbhdPyT

barbeau, 2021-09-27:
>Using a unofficial ROM gives more options of API that where not present before but API are there but not used? I there anything the community can do as aftermarket support?

The low-level drivers that interface with the hardware need to come from the OEM because these are typically a black box. The output of these drivers is then passed into Android. But, if the inputs/outputs don't line up correctly, you can see strange results, ranging from things crashing to interpreting data incorrectly. Without understanding the black box, it's hard to "fix" things.

>Would you want the screen shots for the unknown number 38 sat? Because I have never seen a blank sat before in the app.

Sure!

```
Hardware Version:
  native internal state: Gnss Location Data:: LatitudeDegrees: 59.6134, LongitudeDegrees: 17.8286, altitudeMeters: 58, speedMetersPerSecond: 0, bearingDegrees: 0, horizontalAccuracyMeters: 3.216, verticalAccuracyMeters: 3, speedAccuracyMetersPerSecond: 0.0860233, bearingAccuracyDegrees: 180, ageSeconds: 1.63239e+09
Gnss Time Data:: timeEstimate: 1483228800000, timeUncertaintyNs: 1.57784e+17, frequencyUncertaintyNsPerSec: 200000
Satellite Data for 134 satellites::
constell: 1=GPS, 2=SBAS, 3=GLO, 4=QZSS, 5=BDS, 6=GAL, 7=IRNSS; ephType: 0=Eph, 1=Alm, 2=Unk; ephSource: 0=Demod, 1=Supl, 2=Server, 3=Unk; ephHealth: 0=Good, 1=Bad, 2=Unk
...
constell: 6, svid:   1, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
constell: 6, svid:   2, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
constell: 6, svid:   3, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
constell: 6, svid:   4, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
constell: 6, svid:   5, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
constell: 6, svid:   6, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
constell: 6, svid:   7, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
constell: 6, svid:   8, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
constell: 6, svid:   9, serverPredAvail: 0, serverPredAgeSec:       0, ephType: 2, ephSource: 3, ephHealth: 2, ephAgeSec:       0
```

So my understanding of the above is that the device is obtaining almanac or ephemeris data (i.e., the rough positions of the satellites) from some source and in theory knows that the above list of satellites exist somewhere in the sky. It says `ephType: 2`, which means `Unknown`, and `ephSource: 3`, which is also unknown. `ephHealth: 2` is also unknown. Typically these values should show you were the ephemeris came from (e.g. decoded over-the-air, from SUPL, from PSDS server) and whether it was almanac or ephemeris, but this data seems incomplete/incorrect.

However, looking through the log I don't see any evidence that the device is actually observing Galileo satellites, which could be a hardware or software limitation of the OEM (even if the chipset normally supports Galileo).

barbeau, 2021-10-07:
>Well my theory that GNSS signals can be unlocked like the aftermarket LTE/NR unlocking is technically correct. 

I've been told this is possible on some devices, but on others the hardware is effectively disabled by the chipset OEM when selling to device OEM. Device OEM would need to pay more to enable certain features (e.g., dual-frequency).

Please let me know what you find!

BurhanBudak, 2022-04-08:
> However, looking through the log I don't see any evidence that the device is actually observing Galileo satellites, which could be a hardware or software limitation of the OEM (even if the chipset normally supports Galileo).

Hi, how is it going. Well I own a new phone that supports Galileo. But you pissed the constell value that equals 6. Its says I. The pastebin says GAL=6

barbeau, 2022-04-08:
@BurhanBudak Hello! My comments at https://github.com/barbeau/gpstest/issues/58#issuecomment-928056654 were about the records marked `constell: 6`. From what I recall, the device seemed to be obtaining partial almanac or ephemeris data about Galileo constellation from some source, but there wasn't any indication that the device was seeing live Galileo satellite signals. Almanac and ephemeris can be fetched over the Internet so presence of that data alone doesn't mean that the device is using Galileo to compute a position.

commit 5d559cde (2016-10-08): Fix #58 - Add Galileo support

* In GpsTestActivity - refactor the legacy listener to it's own instantiation so it's encapsulated in a method to switch between legacy GpsStatus listener and new GnssStatus listener
* Add GnssStatus.Callback events to fragments for Android 7.0 and up, make it work for both legacy Pre-N (6.0.1 and lower) devices as well as Android 7.0 and up
* Change "PRN" text to a more generic "ID"
* Add PRN range for Galileo for Pre-N (6.0.1 and lower) - based on early Aquaris X5+ - see https://github.com/barbeau/gpstest/issues/58#issuecomment-252235124.
* Add SNR for Android  [... cut]
commit 48bf9703 (2016-10-10): Merge pull request #59 from barbeau/gnssListener

Fix #58 - Add Galileo support