barbeau/gpstest #76: C/N0 range doesn't seem to be right (Android 7.0)

barbeau, 2016-11-28:
**Summary:** 

On Android 7.0, the C/N0 range we're using for the sky view color fills don't seem to be right.

I used the following reference from AOSP https://github.com/android/platform_frameworks_base/blob/master/location/java/android/location/GnssMeasurement.java#L452:

> Gets the Carrier-to-noise density in dB-Hz.
Typical range: 10-50 db-Hz.

However, this doesn't seem right.  On a LG G5 with Android 7.0 (N), the values seem to be in the same range as SNR from Android M, which was 0-30:

[image]

Our current range of 0-50 leaves us with all red sats in Sky view:

[image]

If you disable the "Use GNSS APIs" option in GPSTest to use the legacy `GpsStatus.Listener` and SNR, here's what the sky view looks like (using 0-30 range):

[image]

C/N0 is being converted to SNR (or SNR to C/N0) for the above screenshots, so they should ultimately represent the same color on the signal quality spectrum.

**Steps to reproduce:** 

Start app with default settings on Android 7.0

**Expected behavior:** 

Show some yellow/green C/N0 values

**Observed behavior:** 

All C/N0 values are always red.

**Device and Android version:** 

LG G5 with Android 7.0

mbavaro, 2016-11-29:
GNSS C/N0 and SNR are beautifully explained here:
http://www.insidegnss.com/node/2397

The correct range for C/N0 is indeed 10-50 dBHz. With good antennas it is not uncommon to see even slightly above 50 dBHz. No commercial GNSS receiver is capable of measurements at C/N0 below 10 dBHz.

In the screenshot above, no C/N0 above 21.3 dBHz can be seen. I would definitely represent all satellites "red" in this case as very few of those measurements are going to be "real". With nominal C/N0 being around 42 dBHz, values of 12 dBHz would require 1 second coherent integration time and I doubt it is possible with low cost temperature-compensated crystals.

barbeau, 2016-11-29:
Thanks @mbavaro!

I found this document, which includes examples - from http://gauss.gge.unb.ca/papers.pdf/SNR.memo.pdf:

| SNC | SNR(db:1khz) | C/N0(db:1Hz) |  |
| --- | :---:|:---:| --- |
| 3 | 6.5 |  36.5 | Very weak signal |
| 5 | 11 | 41 | | 
| 10 | 17 | 47 | | 
| 20 | 23 | 53 | | 
| 30 | 26.5 | 56.5 | | 
| 40 | 29 | 59 | Very strong signal |

It seems then that 0-30dB is a reasonable range for SNR?

mbavaro, 2016-11-30:
SNR is very uncommon for GPS and it shall be negative as signals are way below the noise floor. My guess is that receivers report always C/N0 but Android incorrectly tags it as "SNR".  It should be easy to verify how in the same conditions, SNR and C/N0 (from different API versions) have exactly the same values.

The range 0-30 seems just to take into account that the antenna inside a phone have gains between -7dBiC and -12 dBiC which will lead to C/N0 hardly over 35-40 dBHz. Note finally that with a simulator one can easily achieve values >50 in the "SNR" indicator.

Summarising, I suggest using the same range for both C/N0 and SNR (as in fact I believe they are both C/N0). A range up to 55 is appropriate with a good antenna, but perhaps one could lower it to 45 considering a bad smartphone antenna, just remembering to cap higher values which will happen when somebody tests the phone using a playback device, a re-radiator or a simulator.

barbeau, 2017-07-11:
Finally getting back to this :).

So looking at the output of C/N0 and SNR values on a LG G5 w/ Android 7.0, it does indeed seem that the same numeric values are being provided for both SNR and C/N0.

Here's how I plan to handle the color shading in the Sky view going forward:

1.  Sky view on Android 7.0 and higher - To determine color shading (red->green), SNR and C/N0 will both use the range of 10 (fully red) to 45 (fully green).  If someone toggles between viewing SNR and C/N0 (using the "Use GNSS APIs" setting), this will result in more intuitive experience where the color shading for a given numeric value remains consistent between SNR and C/N0.
1.  Sky view on Android versions below 7.0 - Nothing will change - color shading for SNR (C/N0 isn't available) will continue to use the range of 0-30 for SNR.

The reasoning for 3) is that I don't want to suddenly change the behavior for users that are accustomed to the current range, especially because we don't have a color-coded legend in the app, and the same range has been used for years.  The 2) change will be more subtle, as I'm guessing few users are switching between SNR and C/N0.

Note that the above ranges only affects the color shading in the Sky view - the raw numbers for SNR and C/N0 will continue to be shown in the UI and logged without any modifications.

@mbavaro Does this sound reasonable to you?

commit ca45cb96 (2017-07-14): Fix #76 - Normalize satellite color shading across SNR and C/N0 for Android 7.0 and higher

* Sky view on Android 7.0 and higher - To determine color shading (red->green), SNR and C/N0 will both use the range of 10 (fully red) to 45 (fully green). If someone toggles between viewing SNR and C/N0 (using the "Use GNSS APIs" setting), this will result in more intuitive experience where the color shading for a given numeric value remains consistent between SNR and C/N0.
* Sky view on Android versions below 7.0 - Nothing will change - color shading for SNR (C/N0 isn't available) will continue to use [... cut]
closed 2017-07-14 by barbeau