barbeau/gpstest #79: Support external GPS via Bluetooth

barbeau, 2017-03-28:
**Summary:** 

From a user in the Google Group at https://groups.google.com/forum/#!topic/gpstest_android/Uwmhvp-glfA:

>Is there any chance that GPSTest will support using external bluetooth GPS?

I'd like to support this.  I assume this means supporting NMEA sentences over Bluetooth serial port profile, although I'd like further clarify from actual users for what devices they'd like to see supported to make sure we can get this right.

**Steps to reproduce:** 

1. Start GPSTest

**Expected behavior:** 

Give me an option to see external Bluetooth-connected GPS in the app

**Observed behavior:** 

There is no way to view external Bluetooth GPS in the app

tzac, 2017-04-15:
I have Stonex S5 external GNSS receiver and android 6.0

barbeau, 2017-04-15:
Thanks @tzac!  I emailed Stonex earlier today to ask for technical specs to connect to this device.  Did you get a GNSS Server APK with your device? If so can you share that?

tzac, 2017-04-15:
I paired my Huawei P9lite mobile and S5 with Bluetooth GPS apk from GG moblab and enabled Mock GPS provider. After that GPS Test showing status from S5 but without sat list

barbeau, 2017-04-18:
So to my understanding the Mock GPS provider should let you see location lat/longs from external GPS in Android apps, but to my knowledge you can't see status information like elevation, azimuth, signal strength or satellite IDs per satellite - that comes from a different Android API.

Have you found any other Android apps that allow you to see a list of satellites from your external device?  If so, what are they?

BTW, my email to Stonex bounced, so I'm not sure how to get more detailed technical information about possible integration.

tzac, 2017-04-25:
Bluetooth GPS app showing all that info on status tab page, and on NMEA tab page all nmea log from external gps

barbeau, 2017-04-25:
Thanks - this is the app?

https://play.google.com/store/apps/details?id=googoo.android.btgps

Looks like it reads a NMEA sentence over Bluetooth serial port profile. I'll see if I can contact developer and get more details.

tzac, 2017-04-25:
Correct

panosmark, 2017-08-30:
Hi this project has external gps support and it is open source!!!!!
https://github.com/platypii/BASElineFlightComputer

barbeau, 2017-08-31:
Thanks @panosmark!  I'll check it out.

barbeau, 2017-08-31:
Here's the Bluetooth management code:
https://github.com/platypii/BASElineFlightComputer/tree/master/app/src/main/java/com/platypii/baseline/bluetooth

sp00n, 2019-09-06:
Is this still on your list?
I'd have a couple of external Bluetooth GPS devices that I'd like to see the app to work with:
- Garmin GLO 2 (unknown chipset)
- GNS 2000 plus (MediaTek MT3333)
- Nokia LD-3W (SiRF Star III)
- Qstarz BT-Q818 XT (MTK II)

Other interesting chipsets would be MediaTek MT3318 and MediaTek MT3329 (used by some Holux devices).

I currently have the Nokia, the Garmin and the GNS 2000 available, although I'm thinking about returning one or both of the Garmin and the GNS 2000, as I'm not 100% satisfied with their accuracy. Which is how I stumbled across your app, to maybe gain some more insight, which unfortunately doesn't seem to work yet.

So if you need any more information I could provide for these devices, let me know, before I possibly return them.

barbeau, 2019-09-07:
@sp00n It's still on my list of features I'd like to add to the app but I just haven't gotten to it yet.

For this and sending NMEA over USB/serial (https://github.com/barbeau/gpstest/issues/308) it looks like I may need to get a hold of an actual device for development.  It's hard to implement something when you don't have anything to test against.

sp00n, 2019-09-07:
Well, you can get a used Nokia LD-3W for around 15€ on eBay these days. Not sure about the US though.

gdt, 2020-11-22:
This perhaps belongs in a separate PR, but in addition to bluetooth it would be nice to support NMEA over TCP sockets.  Vespucci has support for this now, to be able to use local rtklib on the phone or an external GPS receiver over wifi.

barbeau, 2020-11-23:
@gdt Could you open a new issue for TCP support and link to this issue? If you have any links to existing tools/code you could share that would help that would be great.

gdt, 2021-06-03:
Hey bot: Despite no activity, this is still valid.

gdt, 2022-04-20:
Hey bot: Despite no activity, this is still valid.

And, here's a device that can provide NMEA over Bluetooth SPP: https://www.sparkfun.com/products/18442

gdt, 2022-11-07:
Hey bot: Despite no activity, this is still valid.

Another device is the Sparkfun RTK Express.    A fairly large number of devices just send NMEA over a bluetooth serial port, so the app connects to SPP (serial port profile).   Another app that can talk to bluetooth GNSS receivers is QField.

abw8353, 2023-04-30:
I'd like to see this have another look. I am having a heck of a time with what seems to be an intermittent issue with an expensive (for me) bluetooth gnss receiver through the "Bluetooth GNSS" app https://github.com/ykasidit/bluetooth_gnss/). GPStest (my go-to app for gnss issues) is only showing the  accuracy plot and histogram, anything else useful is blank. I am attaching an nmea log (from an obstructed sky) in case that helps to see what can be collected from the gps (the accompanying csv show position perhaps a bit more consistent than plausible), for whatever that might be worth.

Another Android app, Locus Map, has an option for switching between internal and bluetooth GNSS. It optionally shows a sky map with signal strength histogram, may be getting more accuracy than is provided by "mock location" which is the primary location sharing channel for apps such as Bluetooth GNSS.

=-=-=
NMEA: 
[2023-04-29_19-34-18_rx_log.txt](https://github.com/barbeau/gpstest/files/11360472/2023-04-29_19-34-18_rx_log.txt)

Locations:
[2023-04-29_19-34-19_location_log.csv](https://github.com/barbeau/gpstest/files/11360471/2023-04-29_19-34-19_location_log.csv)

gdt, 2023-04-30:
The mock location bit isn't about accuracy.    That is just the way location is injected so that the rest of the phone uses the external location.    You could also use QField.    However, what you really should do is take this to someplace that is outdoors with a good sky view and see how it behaves there after being powered up for 15 minutes.

abw8353, 2023-04-30:
As a workaround, supporting an option for input from an NMEA log file would be quite useful - https://github.com/barbeau/gpstest/issues/641

abw8353, 2023-04-30:
>...  what you really should do is take this to someplace that is outdoors with a good sky view and see how it behaves there after being powered up for 15 minutes.

Yes, thank you, I do appreciate the response, but in this instance I have spent several hours doing that over the last few days, with often suspect or worse behavior. I'm returning the external GNSS receiver in this instance, but I would like to be able to work with better tools (eg GPStest) in the future.
At one point I seemed to see lower resolution location data, and wondered if the mock locations were being truncated or rounded. However, others report realtime NTRIP working fine with appropriate receivers, apparently through mock locations, so my concerns in that regard would have faded if I had stopped to think a bit.  

My primary use case is under obstructed sky (rough terrain, dense tree canopy), with relatively rapid TTFF needed. A 15 minute delay at each of multiple sites is not acceptable. 
Accuracy after 15 minutes under an open sky can be good, goofy fun but doesn't budget out.

Before I subject myself to another several-day testing process, I need to enhance my procedures and tools. If GPStest supported Bluetooth receivers either directly or via an option to input from NMEA log files would be one such step forward.

[Also I'd like a tool that would translate my .txt file NMEA log into something resembling my .CSV location log (files in my first comment).]

gdt, 2023-04-30:
I said 15 minutes so that you'd be sure to have ephemeris for all satellites.   If you are in an open field and you don't get a position within a few minutes that is stable to better than 5-10m then something is quite wrong.

I don't see any reason mock locations would have granularity bigger than about a meter, and probably it is much finer.  Degrees to 8 decimal places is about a mm.

Check out gpsd, which you can run on a computer (vs phone), and will process the NMEA and you can then log with gpspipe -w, and then feed to gpsprof, which will give you the analysis that I think you are looking for.

barbeau, 2023-11-16:
Not stale

jcinternetstuff, 2024-03-20:
Bloody stupid bot!

sp00n, 2025-04-24:
Do we want to let it go to sleep?

barbeau, 2025-04-24:
Let's keep it open