barbeau/gpstest #81: F-droid availability

JonnyTech, 2017-04-15:
Are you able to make gpstest available on [f-droid](http;//f-droid.org) for those of us who do not use Google apps?

JonnyTech, 2017-04-15:
Or maybe @IzzySoft can add the apk from https://github.com/barbeau/gpstest/releases to his repo?

JonnyTech, 2017-04-15:
Please ignore me, it is already in f-droid, sorry for the noise.

IzzySoft, 2017-04-15:
Yupp. In my repo, as I indeed already have added the `.apk` from `releases/` back in October last year :smile_cat:

JonnyTech, 2017-08-15:
@IzzySoft, your [repo](https://apt.izzysoft.de/fdroid/index/apk/com.android.gpstest) still points to v2.1.8 - can you change it to the [latest release v2.1.11](https://github.com/barbeau/gpstest/releases/latest)

barbeau, 2017-08-15:
@IzzySoft Also, for "Anti-features", v2.1.4 and higher of the app does not require Google Maps in order to install/run, although you do need it to view the map on the "Map" tab.

IzzySoft, 2017-08-15:
@JonnyTech Sorry for that! Updater faulty. Considered 2.1.8 newer than 2.1.11 (string comparison on the version. I know, not optimal – but you have to compromise with those many different tagging variants used). Updated it manually. Thanks for pointing it out!

@barbeau Sorry again. But my library analyzer says it uses GMS *and* Maps – and users expect me to point that out (yes, I had reports on where I missed it), so it needs to stay. Seeing the AntiFeature, one can check the library section and decide. GMS is not a show-stopper for many, and neither is Maps (in fact, for me both is covered by microG).

barbeau, 2017-08-16:
@IzzySoft Google Maps v2 is part of Google Play Services (GMS) library.  It's fine with me if the AntiFeature notification needs to stay, I just wanted to clarify that even without Google Play Services GPSTest can run on the device.  I confirmed with the latest version v2.1.11 that it can run on an Amazon Fire Phone, which doesn't have Google Maps or Google Play Services.

IzzySoft, 2017-08-21:
@barbeau I could add that to the description if you wish. Also, even if that were not the case, it certainly would work fine with microG :wink:

IzzySoft, 2017-08-24:
@barbeau before I add that to the app's description:

1. if it runs fine without GMS, maybe you can leave GMS out completely? If it's just for maps, AFAIK there is [an OSM lib](https://github.com/osmdroid/osmdroid/) you could use instead.
1. if you cannot leave it out for some reason: what are the limitations on a device that has no Google Apps, so I can point that out (e.g. "Map" tab will remain empty)?

barbeau, 2017-08-26:
1. IIRC I couldn't do the same realtime map orientation and tilt based on accelerometer and gyroscope with osmdroid ad I could with Google Maps API v2. Also, Google Maps API v2 has 3d building models in the map, which I wanted to show to help demonstrate the impact of nearby buildings on GNSS signals. 
2. Yes, limitation is that the map won't appear on the Map tab.

IzzySoft, 2017-08-29:
@barbeau thanks! Updated the description of the app (should become visible this evening). Using "Google Maps API v2" means the standard microG installation should cover this as well (just to let you know).

barbeau, 2018-04-09:
@IzzySoft Thanks for hosting GPSTest in your repo. I'd like to explore getting GPSTest into the main F-droid repo to make it easier for users to access. 

I was looking at the submission process and it looks like it requires config files to be created. Is this something you already have as part of the hosting on your repo? What next steps would you suggest for transitioning the app to F-droid?

IzzySoft, 2018-04-20:
As I only serve the binaries, I've not setup the build part required by F-Droid. So next steps would be:

1. You'll need a flavor without proprietary dependencies (here: GMS)
1. Open a [request for packaging](https://gitlab.com/fdroid/rfp/issues/) (I was just about to do that, but no chance with GMS still in there)

Someone of the F-Droid Team will guide you through any additional steps which *might* be required – i.e. "if any".

paulle, 2018-06-01:
It would be very nice to see gpstest on f-droid. I'm always first looking in f-droid to find an app fitting my needs and if I don't find nothing I'm looking in playstore.

barbeau, 2018-06-16:
Starting with v3.1.0 (https://github.com/barbeau/gpstest/releases/tag/v3.1.0), there is now a `osmdroid` build flavor that uses OSM Droid in place of Google Maps.  I'll be submitting the `osmdroid` build flavor to F-Droid shortly.

barbeau, 2018-06-16:
F-Droid request for packing at https://gitlab.com/fdroid/rfp/issues/627.

paulle, 2018-07-11:
GPSTEST is on f-droid now (v3.1.1). I installed it from there yesterday.   Thank you!

barbeau, 2018-07-12:
You're welcome!  Looks like its now listed at:
https://f-droid.org/packages/com.android.gpstest.osmdroid/

closed 2018-07-12 by barbeau

IzzySoft, 2018-07-12:
Congrats! Removing it from my repo then to avoid confusion (signature mismatch with the one from the official repo).

barbeau, 2018-07-12:
@IzzySoft sounds good! Thanks for your help in this process, and thanks for hosting it in your repo in the interim.

IzzySoft, 2018-07-12:
Always glad to help!