barbeau/gpstest #92: Add flag for GAGAN

kanchudeep, 2017-09-02:
[GPS Aided GEO Augmented Navigation](https://en.wikipedia.org/wiki/GPS_Aided_GEO_Augmented_Navigation) (GAGAN) satellites (PRN 127, 128, 139) in the listing are shown without a corresponding country flag.

Request add flag of India in the listing to ease identification of the satellite(s)/constellation.

Note: Refer the attached screenshot and notice no flag with PRN 127.

[image]

barbeau, 2017-09-02:
Thanks @kanchudeep! Yes, I can add Gagan. What mobile device (make and model) is the screenshot from, and what is the Android version?

kanchudeep, 2017-09-03:
Welcome @barbeau...
The above screenshot is on a OnePlus 5 running Nougat (7.1.1).
GAGAN satellites are also listed on my other following devices:

1. Asus/Google Nexus 7 (2013) running Marshmallow (6.0.1).
2. Samsung S2+ running custom AOSP Marshmallow (6.0.1) based ROM.
3. Asus/Google Nexus 7C (2012) running Lollipop (5.1.1).

If you want I can also send you the ADB logcats...

Also, I'm in India and have observed these satellites in both Northern and Eastern part of the country.

barbeau, 2017-09-04:
@kanchudeep Can you please send screenshots for the other devices too?  If you could turn on NMEA logging via "Settings->NMEA" and send logs that would be great.

One question - how do you know that the PRN range is 127, 127, and 139?  Have you seen all three of these on your devices?

kanchudeep, 2017-09-04:
@barbeau, I'll get the screenshots, NMEA logs and attach.

As for PRN refer:
[image]
(Original page http://www.isac.gov.in/navigation/gagan.jsp)
GSAT-8 is marked as PRN127 and GSAT-10 as PRN128.
GSAT-15 (captioned as in-orbit spare) is not marked with any PRN No.

https://www.pulsesat.com/satellites/s/02d3cd67-dff7-4517-935c-9201cdbb3909/ and https://www.celestrak.com/NORAD/elements/sbas.txt states the PRN as 139.
There are some conflicting sources which state the PRN as 132, but I'm more inclined towards 139.

I'll try and see if I get either PRN 132 or 139. I do remember catching 127 and 128 distinctively (in fact just before taking the screenshot in the first post).

I'm the developer of an Android navigation app [Deesha](https://play.google.com/store/apps/details?id=com.deeppradhan.deesha2) and as one of it tools it gives information on GNSS satellites, so I have been researching with different devices and GNSS satellites. So most of this information on GAGAN is based on that research/experience.

barbeau, 2017-09-23:
@kanchudeep If you're just able to provide screenshots of all the GAGAN PRNs you can see from your device, that should be enough for me to implement this (don't worry about logs, etc).  I'd like to get it out in the next version.

kanchudeep, 2017-09-24:
@barbeau, sorry I have been busy...

The GAGAN SVs seen on OnePlus 5:
[image]

The GAGAN SVs seen on Nexus 7 (2013):
[image]
**Note**: The Nexus 7 (2013) seems to report NMEA IDs instead of PRN. You need to add 87 to get the PRN from the NMEA IDs.

I have not been able to observe either PRN132 or PRN139 of GSAT-15.

closed 2017-11-15 by barbeau

barbeau, 2017-11-17:
@kanchudeep I've published a new release including GAGAN and the Indian flag (v2.2.0) to the beta channel, which you can either sign up for [here](https://github.com/barbeau/gpstest/blob/master/BETA_TESTING.md#register-as-a-beta-tester) or just download the APK directly from Github at https://github.com/barbeau/gpstest/releases/tag/v2.2.0.

Could you please give this a try and let me know if you see the Indian flag now?

Screenshots welcome :).

kanchudeep, 2017-11-24:
Greetings...

Sorry, I didn't get time to check earlier.

Attaching a screenshot.
It is displaying Indian flag correctly for GSAT-8/PRN 127.

Regards...
दीप प्रधान

barbeau, 2017-11-24:
@kanchudeep Thanks! Could you post screenshot again? It didn't attach last time.

kanchudeep, 2017-11-24:
Greetings...

Screenshot attached again.

One issue I think I might have noticed. The last flag seems to be cropped from the bottom. I see this with the Indian flag in this screenshot and also noticed with some other flags.

Regards...
दीप प्रधान

barbeau, 2017-11-25:
If you're attaching via email I don't believe that works - you need to add from desktop web view.

commit 34637d6c (2017-11-15): Fix #92 - Add support for GAGAN (Indian GNSS)