barbeau/gpstest #95: Lighten SBAS flags?

goldfndr, 2017-11-17:
Could the country flags for SBAS satellites be lightened to help distinguish them from non-SBAS satellites?

There was a Firefox extension that added a country's flag onto the end of the address bar that represented the country from which the IP address for the website originated. I think it might've been "Country Flag". It had an option of "palette" which could be "light" or "dark", that would pick one from two sets of flags, one somewhat lightened and one possibly darkened.

I think that this approach, dark and light flags, would be an easy way to readily distinguish the SBAS satellites from the non-SBAS satellites, giving more information without needing to alter the table layout.

goldfndr, 2017-11-17:
  Alternatively, maybe add "S" to the flags column for SBAS?

barbeau, 2017-11-19:
Thanks - yes, I'll think about this a bit more. There are multiple dimensions of information here I've been trying to figure out to represent clearly and succinctly - related to #90.

goldfndr, 2018-09-20:
https://flagfox.net/viewtopic.php?t=553&start=30#p1856 includes an image of the dark and light flags.

barbeau, 2018-09-20:
From https://github.com/barbeau/gpstest/issues/214#issue-361347438:

> I believe the current labels for SBAS are not meaningful.
> 
> For example for EGNOS the system is identified with the Luxembourg and UK flags, while that only the nationality of the commercial communications satellite used. Luxembourg and UK do not operate a SBAS system, the EU does.
> 
> Also for WAAS you picked just Galaxy 15 and Anik, and list them separately, but for Gagan you list all together.
> 
> It's much more didactic if you merge Anik and Galaxy 15 as WAAS, with Canada and US flag, and merge all the EGNOS SVs as EGNOS, with the EU flag.
> 
> Btw, for Galileo you used the project logo (not EU flag), but for the other projects you choose the country flag.
> 
> Sorry to be pedantic, I can't avoid it, but I think it's something easy to get consistent!
> 
> Thanks!

My reply:

thanks for the feedback! I agree that the presentation of SBAS could be improved. It's been largely done ad-hoc to-date as SBAS started appearing on Android device but I'm in favor of revising it. I'm going to do a little more research and will come up with a plan.

closed 2018-10-01 by barbeau

kimollivier, 2019-01-19:
Australia and New Zealand have SBAS as a trial, now permanent. Do you have to actively search for it? Is it not automatic? Can GPS test be used to mock the internal app if SBAS is enabled?

barbeau, 2019-01-20:
On Android 7.0 and higher you would see a record for these SBAS satellites show up in the bottom card but they just wouldn't have a flag next to them. If you spot these please share screenshots and I'll add flag images!

commit b07de84e (2018-10-01): Fix #95 - Update SBAS presentation (#217)

* Show SBAS in Status screen as a new card under the GNSS card
* Refactor all individual satellites to their SBAS constellations
* Refactor model classes to model package, add Kotlin support
* Update Gradle plugin and Gradle
* Bump buildToolsVersion, update Travis
* Remove unused strings in translations
* Update to compileSdkVersion 28
* Migrate to AndroidX
* Update README, FAQ, and Help strings with new SBAS constellations
* Remove unused variables and resources