barbeau/gpstest #96: Add country flags to Help or Readme?

goldfndr, 2017-11-17:
Could the country flags be added to the Help or the Readme?

Some people might not have the looks of particular country's flags memorized (yet). Adding the country flags to the Help or Readme, along with the country name and the name of the GNSS and SBAS constellation each represented, would help educate people rather than having them try to somehow look up the images. It would also be a method to "show off" all the satellite sources the installed version of GPS Test supports.

barbeau, 2017-11-19:
Thanks, I'll look at this. Definitely a good idea but more cosmetic items have just been towards the bottom of the TODO list.

barbeau, 2018-07-12:
Flags and names are now shown in the Legend in the Sky view, so I'm closing this.

closed 2018-07-12 by barbeau