barbeau/gpstest #100: PRNs 48 and 51 not showing flags

goldfndr, 2017-11-18:
**Summary:** 

The rows on the Status screen corresponding to PRNs 48 and 51 don't display flags.

**Steps to reproduce:** 

Launch the app (2.2.0) in the USA, wait for the satellite to be received.

**Expected behavior:** 

Display a flag for PRNs 48 and 51. I thought I saw that in 2.1.11 but now I'm not sure. Let me know if you'd like me to try a downgrade.

**Observed behavior:** 

PRNs 48 and 51 don't have flags, as if they're unrecognized. (Also, I find it odd that they can be Used even though they don't have a displayed Azimuth value.)

**Device and Android version:** 

Moto G (2013) with Android 4.4.4, stock VZW ROM with Xposed and a few Xposed modules

**Screenshots:** 

Available upon request. I'm not spotting a method in ForkHub to attach a file.

barbeau, 2017-11-19:
@goldfndr thanks for reporting this! Could you include a screenshot? You'll probably have to visit the website in Desktop view to add the screenshot.

Yes, going back to Android 4 support for WAAS and other sats will be very spotty, so you might see strange things.

goldfndr, 2017-11-22:
Sure. Here you go. I have also, in the past, detected NMEA ID 46 with SatStat, it's probably only a matter of time before I can detect it with GPSTest.

 [image]

goldfndr, 2017-11-22:
It looks like it'll be pretty trivial to add to https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/java/com/android/gpstest/util/GpsTestUtil.java given the subtraction of 87 from the GNSS (Android 7+) values.

barbeau, 2017-11-22:
Thanks!  I'll mostly likely just add this to the legacy getGnssType() method:
https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/java/com/android/gpstest/util/GpsTestUtil.java#L49

...which is for Android 6.0.1 and lower.  I've been trying to stick to actual observations on real devices and avoiding speculative additions, so I'll wait to add these on Android 7+ until someone reports them on actual devices.

goldfndr, 2017-11-22:
You didn't wait to add these for 7.0, they're already there, GALAXY_15 and ANIK!

barbeau, 2017-11-22:
@goldfndr Ah, right, of course. Sorry, too much multitasking :).

goldfndr, 2017-11-28:
Well, it looks like I can stop checking for 46 to reappear. [It was decommissioned on Nov 9.](http://www.nstb.tc.faa.gov)

So now I'm waiting for [44](http://continuouswave.com/forum/viewtopic.php?f=9&p=16016#p15325) to show up.

goldfndr, 2017-12-19:
Well, I don't know how it happened, but... somehow 46 came back online tonight. [See screenshot](https://photos.app.goo.gl/UZKim5OpWH9Qqyt02).

barbeau, 2017-12-19:
Interesting...

goldfndr, 2018-01-06:
46 might (again) be moot. Since around the start of the year (unsure between Dec 27 and Jan 3), I can no longer detect 46, just 48 & 51.

Since I'm not there, I can't speak for NMEA 33 (Inmarsat 3-F2, part of the EGNOS SBAS).

goldfndr, 2018-01-13:
And now 46 is back today, even with a bit of rain, even though it's still supposedly decommissioned.

goldfndr, 2018-01-31:
It finally happened. 46 was usable (and used!) for the SBAS. [See screenshot.](https://photos.app.goo.gl/iV8daHmSlSdAmZyG2) And you can see that my DOP values are quite low (lower than usual) because of it.

barbeau, 2018-02-01:
To mirror part of my post at https://github.com/barbeau/gpstest/issues/106#issuecomment-362324992 here:

For 46 (133), http://www.navipedia.net/index.php/WAAS_Space_Segment says:

Satellite Name & Details | NMEA / PRN | Location
-- | -- | --
Inmarsat 4F3 (AMR) | NMEA 46 / PRN 133 | 98°W

*  Inmarsat 4-F3 - https://nssdc.gsfc.nasa.gov/nmc/spacecraftDisplay.do?id=2008-039A - Funding agency is Inmarsat

barbeau, 2018-02-01:
I'm moving 46 (133) to it's own issue at https://github.com/barbeau/gpstest/issues/108.

closed 2018-02-05 by barbeau

goldfndr, 2018-04-25:
Even though 46/133 was decommissioned in early November, it came back, and I've been seeing it fairly regularly. Less than an hour ago, it was even Used in the solution (AEU flags), something I hadn't witnessed in a couple of months (although I'll admit that I've been running GPSTest sporadically rather than daily lately). It's showing up in the [WAAS Satellite Corrections page](http://www.nstb.tc.faa.gov/RTData_WaasSatelliteData.htm).

I'm still not seeing 44/131 yet, even though [others are](http://continuouswave.com/forum/viewtopic.php?f=9&p=16016#p18624). Any day now...

goldfndr, 2018-07-15:
Just an update: I'm **still** seeing 46, regularly getting AE flags for it, and just now had U as well. Still not seeing 44 yet.

ndoggac, 2018-07-25:
@goldfndr from comment in #103 .  Just the thread my google searching brought me to, didn't realize this thread existed.  Seeing PRN133 is expected, as the new GEO currently in IOT is broadcasting on PRN133.  However, it is in test mode, so the WAAS algorithm on chip shouldn't be utilizing the data stream of corrections.  Not sure if/how this is reflected in the satellite status information.

Wiki I was referring to was the previously linked WAAS space segment wiki section.  

The following FAA link discussed the next coming GEO (SES-15 @ 129W) on PRN133.  https://www.faa.gov/about/office_org/headquarters_offices/ato/service_units/techops/navservices/gnss/library/satnav/media/SatNavNews_Spring_2017.pdf

goldfndr, 2018-07-26:
The linked PDF's only mention of "133" is "Intelsat Galaxy 15 at 133 ̊ West". Could you link your source for SES-15 getting PRN133, and quote the relevant sentence verbatim? Or, if an image, which image? That's needed for updating the English Wikipedia's WAAS space segment section.

> Seeing PRN133 is expected

Are you asserting that any of the transmissions from the dates above:  
* December 18
* January 13
* January 31  

came from SES-15 and not Inmarsat 4-F3? If so, which?

commit 31871125 (2018-02-05): Fix #100 - Flags for WAAS IDs 48 & 51

This change displays the flags for WAAS IDs 48 and 51 for Android versions below 7.0.

While I was at it, I moved NAVSTAR to the top, both for numerical order and to have it first in the decision tree instead of last (as it tends to be pretty popular for older chipsets).
commit 944369dc (2018-02-05): Merge pull request #110 from goldfndr/patch-2

Fix #100 - Flags for WAAS IDs 48 & 51