barbeau/gpstest #102: BQ Aquaris X5 Plus does not support logging

johnathn, 2018-01-04:
**Summary:** 

BQ Aquaris X5 Plus does not support tracking of GNSS Navigation Messages

**Steps to reproduce:** 

I downloaded the app from the playstore, opened the app, went to settings menu, scrolled down, checked the checkbox for logging navigation messages and meassurements and return to the main activity, I get the two toasts, telling it does not support tracking.

**Device and Android version:** 

BQ Aquaris X5 Plus with Android 7.1.1

barbeau, 2018-01-04:
@johnathn Unfortunately this is a limitation of your device.  You can see what devices currently support Navigation Messages in this table:
https://github.com/barbeau/gpstest/blob/master/FAQ.md#what-android-70-devices-support-logging-raw-pseudorange-and-navigation-messages

I also intend to add devices to the FAQ from this page (related to https://github.com/barbeau/gpstest/issues/83):
https://developer.android.com/guide/topics/sensors/gnss.html#supported_devices

I'll leave this issue open as a reminder to myself to add BQ Aquaris X5 Plus to that table in the FAQ.

barbeau, 2018-01-05:
@johnathn Can you please test logging NMEA and raw measurements too to see if those are supported on your device?

commit ec95b2fb (2018-01-05): Fix #102 and #83 - Update device logging table

* Google seems to be keeping their list up-to-date, so just add info for our own observations that aren't on Google's page in this repo
closed 2018-01-05 by barbeau