barbeau/gpstest #107: Some "Help" list items have weird spacing

barbeau, 2018-02-01:
**Summary:** 

If you open the Help menu via 3 dots menu button and then "Help", some of the list items have strange spacing.  For example, in the below screenshot from a Samsung Galaxy S8+ w/ Android 8.0 beta the GNSS list looks fine but the "Acknowledgments" section is off:

[image]

**Steps to reproduce:** 

1. Launch app
1. Tap on 3 dots for overflow menu
1. Tap "Help"
1. Scroll to bottom

**Expected behavior:** 

List items should appear evenly spaced on one line like the GNSS list

**Observed behavior:** 

Some sections like "Acknowledgments" spill onto 2nd lines from bullets

**Device and Android version:** 

Samsung Galaxy S8+ w/ Android 8.0 beta

**Screenshots:** 

See above

goldfndr, 2018-02-02:
There's also weird spacing above and to the left of the Sky View header – it should have the same spacing as the Acknowledgements header.

barbeau, 2018-02-02:
Thanks @goldfndr!  I believe I just fixed that too.

commit eaebdb3f (2018-02-02): Fix #107 - Fix strange formatting of Help text
closed 2018-02-02 by barbeau