barbeau/gpstest #108: WAAS IDs 46 not showing flags

barbeau, 2018-02-01:
**Summary:** 

First mentioned at https://github.com/barbeau/gpstest/issues/100#issuecomment-346302124, more at https://github.com/barbeau/gpstest/issues/106#issuecomment-362324992.

PRN 46 has been visible on an Moto G (2013) with Android 4.4.4, stock VZW ROM with Xposed and a few Xposed modules.

[image]

For 46 (133), http://www.navipedia.net/index.php/WAAS_Space_Segment says:

Satellite Name & Details | NMEA / PRN | Location
-- | -- | --
Inmarsat 4F3 (AMR) | NMEA 46 / PRN 133 | 98°W

*  Inmarsat 4-F3 - https://nssdc.gsfc.nasa.gov/nmc/spacecraftDisplay.do?id=2008-039A - Funding agency is Inmarsat

As mentioned in https://github.com/barbeau/gpstest/issues/106#issuecomment-362324992, it's not immediately clear to me what flag/icon should be used to represent this satellite.  Maybe just the Inmarsat icon?

**Steps to reproduce:** 

Open app on device in view of this satellite

**Expected behavior:** 

A flag should be shown

**Observed behavior:** 

No flag is shown

**Device and Android version:** 

Moto G (2013) with Android 4.4.4, stock VZW ROM with Xposed and a few Xposed modules

**Screenshots:** 

See above

barbeau, 2018-02-10:
As mentioned in https://github.com/barbeau/gpstest/issues/106#issuecomment-363187313, Inmarsat is a British company based in London, so we'll use the UK flag.

closed 2018-02-16 by barbeau

goldfndr, 2018-02-21:
Here's proof that it works. (not that you needed it)

[image]

barbeau, 2018-02-21:
@goldfndr Awesome, thanks for sharing!  BTW, are you in the Beta testing group?

https://play.google.com/apps/testing/com.android.gpstest

I've been working on pulling together a new user interface based on material design, which paves the way for adding more functionality to the UI (logging, etc.) via a navigation drawer - implementation of WIP is currently in https://github.com/barbeau/gpstest/tree/oba-drawer.

I hope to do a new beta release of this very soon and it would be good to get feedback of it working on your device.  Initial beta release will likely just be the light theme, but as I mentioned in https://github.com/barbeau/gpstest/issues/112 I want to support light and dark themes.

barbeau, 2018-02-21:
To get the beta update you'd need to uninstall custom build and download from App Store.  Either way works, and if you're monitoring the Github repo you should get a ping when I cut a release.

Yes, the goal is to get it to look the same across all supported Android versions, including pre-Lollipop.  So testing on your KitKat device would definitely help.

goldfndr, 2018-02-21:
Ok. Wow, that's a lot more commits than I expected, you've been busy!

commit d32c5bb6 (2018-02-16): Fix #108 - Add INMARSAT 4F3 (SBAS)