barbeau/gpstest #117: Use 24 hour format if set in system preference

goldfndr, 2018-02-25:
**Summary:** 

My phone is set to 24 hour time but GPSTest only displays 12 hour time.

**Steps to reproduce:** 

Launch app, go to Status view.

**Expected behavior:** 

Display 24 hour time in upper right table cell.

**Observed behavior:** 

12 hour time (with "AM" or "PM" is shown.

**Device and Android version:** 

Moto G (1st generation), Android 4.4 KitKat, rooted with some Xposed modules

**Screenshots:** 

Every screenshot I've uploaded is an example of this.

**Attempt to fix:**

I'd expected that changes to GpsStatusFragment.java of adding `import android.text.format.DateFormat;` and changing the mDateFormat assignment to:

```java
    SimpleDateFormat mDateFormat = new SimpleDateFormat(
            DateFormat.is24HourFormat(getContext()) ? "HH:mm:ss" : "hh:mm:ss a");
```
would do it, but I'm unable to figure out what to use for the *context* parameter (or if an intermediate variable is required); the code above crashes.

barbeau, 2018-02-26:
Thanks @goldfndr for opening this issue!  Lint's been bugging me about this but I'm always in the middle of another change and then I forget about it.  I'll fix this - I think something like this should work:

`SimpleDateFormat mDateFormat = (SimpleDateFormat) SimpleDateFormat.getTimeInstance();`

...if you want to test it.

If you use `getContext()` in the body of the Fragment I believe it will crash because the Fragment isn't attached to the Activity yet - you'd have to move the initialization to another part of the Fragment lifecycle where it has access to the Context (e.g., `onAttach()`).  But, I think the above should work and it's simpler, so let's go with that.

barbeau, 2018-02-27:
Turns out what I put above won't work as it doesn't provide seconds.

And, we can't use [`DateTimeFormatter.ISO_LOCAL_TIME`](https://developer.android.com/reference/java/time/format/DateTimeFormatter.html#ISO_TIME) because it's only API level 26 and higher.  We could use [Android Three Ten Backport](https://github.com/JakeWharton/ThreeTenABP), but it seems like overkill to add a dependency just for this.

So, in summary, @goldfndr I think I like your solution the best - my original concern was that this didn't take locale into account, but looking at [the implementation](https://android.googlesource.com/platform/frameworks/base/+/0e2d281/core/java/android/text/format/DateFormat.java#125) it looks like it does.

To avoid the issue with the Context, and keep the initialization in the body, we can use `Application.get().getContext()`.  I'll fix this.

commit d03948c2 (2018-02-27): Fix #117 - Show 24 hr time format if preference or locale is set
closed 2018-02-27 by barbeau