barbeau/gpstest #130: Numbers don't use commas instead of periods based on locale

barbeau, 2018-03-22:
**Summary:** 

In the app, the following fields don't change periods to commas in numbers based on locale:
* PDOP
* C/N0
* Elev
* Azim

For example:

[image]

**Steps to reproduce:** 

1. Change to a locale where commas are used in place of periods in numbers (e.g., Swedish - `sv_SE`)
1. Run the app and view the Status screen

**Expected behavior:** 

Show me commas in numbers in place of periods

**Observed behavior:** 

In the following fields periods are still shown:
* PDOP
* C/N0
* Elev
* Azim

**Device and Android version:** 

Visible on a Galaxy S9+ screenshot w/ Android 8.0 shared by user

closed 2018-03-22 by barbeau

goldfndr, 2018-03-22:
Would it make sense to simply drop the ",0" or ".0" at the end if all the values end the same way? In every screenshot I've seen, they all were integers, which makes me think that adding tenths make it appear to be more accurate than it actually is.

barbeau, 2018-03-22:
@goldfndr That's a good question - I've observed the same.  However, they are floating point values coming from the Android APIs, so I'd prefer the leave open the possibility that some devices may provide further precision.

goldfndr, 2018-03-22:
Do you think it's possible that some future device might have tenths (or higher) accuracy but have some readings in which all C/N0 and Elev and Azimuth values are integers?

barbeau, 2018-03-22:
Good question - I'm not sure.  Since these values come from native code they will differ by OEM and chipset, so I'd expect to see a variety of formats and values.

goldfndr, 2018-03-22:
Er, just Elev and Azim – I've observed tenths on SNR.

goldfndr, 2018-03-22:
If one has just 3 satellites with elevation and azimuth, there's a one in a million chance of them having zero tenths when they'd ordinarily have non-zero tenths. Divide probability by 100 for each additional satellite.

So, it'd be quite safe to check that they all are integers (==Math.rint or zero for modulo 1 or whatever) and not display tenths if so.

>they are floating point values

A nearly identical argument could be made that they should be displayed with 6 decimal digits instead of 1, as that's the approximate resolution in the 0-360 range.

Come to think of it, there's probably a way to determine what location accuracy would be needed to require tenths of elevation/azimuth degrees. 3 meters wouldn't trigger it, but perhaps 0.5 meters (or less) would.

commit 72ccb3ed (2018-03-22): Fix #130 - Format decimals properly for locales

* PDOP
* C/N0
* Elev
* Azim
* Carrier frequency