barbeau/gpstest #135: Proposal for logo new logo design

jbeguna04, 2018-03-28:
Hi good day! I would like to ask for your permission to propose a new designs of logo in your application/project. This issue is not saying that your current logo is unsatisfactory, I would just like to propose new designs. Hoping for your positive response Thank You!

barbeau, 2018-03-28:
@jbeguna04 I'm always open to new ideas, although I just put together a new logo for the v3 release that is rolling out to production now.  You can opt-in to the beta testing group to see it on your device at:
https://play.google.com/apps/testing/com.android.gpstest

Here's a large version of it:
[image]

jbeguna04, 2018-03-28:
Thank you for the quick reply. I understand that

barbeau, 2018-03-28:
Sure!  I'll close out this issue for now just for housekeeping, but if you really want to put something together feel free to post proposals on this thread.  I'd definitely welcome any other UI/UX design suggestions for the app itself as well, but please open new issues for those.

closed 2018-03-28 by barbeau