barbeau/gpstest #141: Visualizing/Downloading GNSS measurements log from Android Studio

binghy, 2018-04-17:
Hi, I'm using GPSTest on a Huawei P10 with Android 7.0. I was interested in logging the raw measurements. I followed all the steps suggested here (USB driver, enabling the log by the app, selecting "no filters" and typing "GpsOutput" in Adroid Studio), but I don't visualize, at least, the measurements on the Android Monitor. This is the screenshot from my laptop. Please, I'm not expert at all with Adroid Studio, so I do not know what all those windows on the top refer to.
[image]

Is something missing from all the steps that I've followed?

barbeau, 2018-04-17:
@binghy Have you enabled USB debugging on your device (which is different than installing a USB driver)?  See https://developer.android.com/studio/run/device.html#developer-device-options

binghy, 2018-04-18:
yes, I did it, even if it struggled me a lot before understanding why USB debugging was going to get disabled every time I quit the settings menu

barbeau, 2018-04-18:
In addition to enabling USB debugging in system settings, did you click "Ok" for the system dialog to allow the laptop to send commands to the device?

>When you connect a device running Android 4.2.2 (API level 17) or higher to your computer, the system shows a dialog asking whether to accept an RSA key that allows debugging through this computer. This security mechanism protects user devices because it ensures that USB debugging and other adb commands cannot be executed unless you're able to unlock the device and acknowledge the dialog.

If you don't do this you won't see anything in Logcat.

binghy, 2018-04-20:
yes, I did it. I clicked ok to allow USB debugging, and then ok once again visualizing the RSA keycode. Moreover, once completed these steps, the tool "HiSuite" shows that the device is connected to the laptop. So, the steps I did in order are:
- open Android Studio (no device connected). Here the options "No Filters" and "GpsOutput" are still present from last time. Device offline.
- plug the USB cable
- enable USB debugging
- click "ok" two times. Now device recognized
- start GpsTest on the Huawei P10
- Settings, enable Measurements log. Going back to the main window the app confirms the success
- nothing on the logcat screen. Here, clicking on the "Device File Explorer", I've noticed that in the log folder, the permission is denied and there's an error. But I do not know if that one should be the right folder to store data, eventually.

Is some of the steps missing to visualize/store data?

I attach some figures to clear all the steps described.

[image]

[image]

[image]

barbeau, 2018-04-20:
>Here, clicking on the "Device File Explorer", I've noticed that in the log
folder, the permission is denied and there's an error. But I do not know if
that one should be the right folder to store data, eventually.

Currently GPSTest doesn't support logging to a file (that's ticketed in
https://github.com/barbeau/gpstest/issues/66 for a future feature), so you
won't find a log file.

It sounds like you've completed all the steps correctly to see the output
in LogCat.  It's possible that some devices don't output Logcat statements
for release builds of APKs (e.g., what you would download from Google Play).

You could test this by uninstall GPSTest from Google Play, and then
building the source code of the project on your machine in Android Studio
and installing that APK on your device.  Instructions are here if you want
to try this out:
https://github.com/barbeau/gpstest#building-in-android-studio

binghy, 2018-04-23:
thank you. I'll try this one and let you know if it works.

Regards,

Francesco

barbeau, 2018-04-25:
I'm closing this issue for housekeeping as it doesn't seem to be related to the app, but let me know your results from running in Android Studio.

closed 2018-04-25 by barbeau