barbeau/gpstest #144: New Logo desing For Gps Test

sametceylan, 2018-04-24:
Hello sir, my name is Samet Ceylan and I am a designer. If you want and let me, I want to design a creative logo for GPS TEST. I want you to know that I will do it for you. I do not want any compensation. If you would like me to design and if you have any particular detail, please contact me.

barbeau, 2018-04-24:
Wow, there are a lot of prospective designers out there ;) (https://github.com/barbeau/gpstest/issues/135, https://github.com/barbeau/gpstest/issues/143).

I just rolled out a new logo in v3 of the app, but thanks for the offer.  Any UX suggestions are always welcome.

closed 2018-04-24 by barbeau

JonnyTech, 2018-04-24:
These offers are mainly from members of sites such as [utopian.io](utopian.io) that reward users for their work.

barbeau, 2018-04-24:
@JonnyTech Ah, interesting.  Thanks for sharing, I hadn't seen Utopian before.