barbeau/gpstest #146: output when sharing position

paulle, 2018-05-23:
**Summary:** 

Summarize your issue in one sentence (what goes wrong, what did you expect to happen)

**Steps to reproduce:** 

How can we reproduce the issue?

**Expected behavior:** 

What did you expect the app to do?

**Observed behavior:** 

What did you see instead?  Describe your issue in detail here.

**Device and Android version:** 

What make and model device (e.g., Samsung Galaxy S3) did you encounter this on?  What Android version (e.g., Android 4.0 Ice Cream Sandwich) are you running?  Is it the stock version from the manufacturer or a custom ROM?  What version of Google Play Services is running on your device?

**Screenshots:** 

Can be created by pressing the Volume Down and Power Button at the same time on Android 4.0 and higher.

paulle, 2018-05-23:
When sharing the postion the output generated by gpstest is a link to googlemaps.
Some people are using other maps like OSM ana a lot of them have maps locally stored on their phone.
For this reason it would be nice to have an additinal output in the following form:
Longtude: xxxxxxxxxxxx
Latitude: xxxxxxxxxxxx
This would it make more comfortable for every receiver of a postion to use it with every map OSM, Here or whatever it is online or offline.

Gpstest is excellent. So far the only app (and I tested a few) which shows the Galileo satellites. Other apps don't do (even when they say having this feature) or only from time to time.

barbeau, 2018-05-23:
Thanks @paulle, I agree that it would be nice to use a standard format for sharing the position that doesn't depend on Google Maps.  I updated the issue description to reflect this.  Glad you like the app!

barbeau, 2018-06-27:
WIP PR at https://github.com/barbeau/gpstest/pull/160