barbeau/gpstest #150: Map should be dark when in Dark theme (Google flavor)

goldfndr, 2018-06-17:
**Summary:** 

The map is light when using a dark theme but should be dark.

**Steps to reproduce:** 

1. Start Google flavor of app.
2. Settings -> Use Dark Theme=ticked.
3. Restart app.
4. Sky (from hamburger menu).

**Expected behavior:** 

Show dark theme map. https://developers.google.com/maps/documentation/android-sdk/styling

**Observed behavior:** 

See light theme map.

**Device and Android version:** 

Moto G (2013), KitKat 4.4.4, stock ROM with some Xposed modules, Google Play Services 12.2.1

**Screenshots:** 

Available upon request.

barbeau, 2018-06-17:
Thanks for the link - I either didn't know or had forgotten you can change the Google base map color via the new styling tools. I'll work on this.

barbeau, 2020-09-09:
Cloud-based styling is now an option as well. although it requires updating to the standalone v3.1 beta SDK:
https://developers.google.com/maps/documentation/android-sdk/cloud-based-map-styling

commit f6dddfd5 (2020-09-09): feat: Support Google Map dark theme

Closes #150
closed 2020-09-09 by barbeau