barbeau/gpstest #152: Sky Avg SNR chart not colored in KitKat.

goldfndr, 2018-06-17:
**Summary:** 

The Sky activity's Avg SNR chart should show a gradient (like in screenshot) insead of a uniformly dark rectangle.

**Steps to reproduce:** 

1. Start Google app.
2. Sky (from hamburger menu).
3. Scroll down as needed until Avg SNR chart is visible.

**Expected behavior:** 

Show a gradient in the interior of the Avg SNR chart.

**Observed behavior:** 

See uniformly dark interior of the Avg SNR chart, regardless of Use Dark Theme ticked.

**Device and Android version:** 

Moto G (2013), KitKat 4.4.4, stock ROM with some Xposed modules, Google Play Services 12.2.1

**Screenshots:** 

Available upon request.

barbeau, 2018-06-17:
Thanks for flagging this! My last KitKat test device died a while back so I'll see if I can repro via emulator. If possible could you share a screenshot?

goldfndr, 2018-06-17:
Sure. Here you go.

[image]
[image]

goldfndr, 2018-06-17:
Also, that tick mark for 15 looks suspiciously non-centered.

barbeau, 2018-06-17:
Thanks! Yeah, not sure what's going on with the 15 tick, will take a look.

goldfndr, 2018-06-18:
In gps_sky_signal_meter.xml, the margins for the tick marks (sky_legend_cn0_\*) appear to be: 2, 25, 44, 63, 2, 98, 117, 136, 2. Granted, those 2s are somewhat aligned with the rectangle, but it looks to me like the space between the last two tick marks (30 and what comes before) is larger than the preceding spaces.

commit 0361229d (2018-06-19): Fix #152 - Use PNG instead of XML drawable for gradient on API < 24
closed 2018-06-19 by barbeau