barbeau/gpstest #153: SNR range incorrect in some Android versions/models

goldfndr, 2018-06-17:
**Summary:** 

The Sky activity's Avg SNR chart should show a range matching the used range of the target device instead of always 0-30.

**Steps to reproduce:** 

1. Start app.
2. Sky (from navigation drawer).
3. Scroll down as needed until Avg SNR chart is visible.

**Expected behavior:** 

The scale should be changed to 15-45 as needed.

Alas, I don't know how to predict what the range should be; I have seen a few screenshots with SNRs of 7 and 8, so there certainly are some devices for which 0-30 is valid. Perhaps a saved preference; if it sees values exceeding 30, bump up Max; if it sees values lower than 10, bump down Min? Yuck.

**Observed behavior:** 

See scale of 0-30, with the balloon readouts on the right margin. The highest average I've today seen was 41.0, but one must have really quick timing and luck to get a screenshot with a high value.

**Device and Android version:** 

Moto G (2013), KitKat 4.4.4, stock ROM with some Xposed modules, Google Play Services 12.2.1

**Screenshots:** 

Can be seen in #152, #108, #100 (all mine), #82, #58 (both 0-30 and somehow higher ranges; see mbavaro screenshot), [Carlos with Aquarius](https://groups.google.com/d/msg/gpstest_android/xulXDk7JqxA/YCgQuqp4BgAJ), [Александр Петрович with Asus me173x tablet](https://groups.google.com/forum/#!topic/gpstest_android/rRn3UChTSr4) (~10-40 range if zeros are invalid). How about [an Android 2.1 competitor](http://apkhere.mobi/download/gpshelper-secure-game-for-android)? And then there's [this YouTube video](https://www.youtube.com/watch?v=s8Jt0mAgQcI) but I don't know how reliable the app used is.

barbeau, 2018-06-17:
Ugh... Yeah, SNR reported by devices is questionable, and on Android 7 and higher its likely just C/N0 passed through as SNR when using the legacy GpsStatus API.

A solution is to keep the two fixed scales and just move to the C/N0 scale when an SNR value exceeds 30. When SNR exceeds 30, do you ever see a value that drops below 15?

I really don't want to get into arbitrary scales based on the values devices spit out because it's messy to manage and likely not correct anyway.

barbeau, 2018-06-25:
Whoops, forgot to update the code in updateSnrCn0Avgs(), reopening.

commit d3f49c25 (2018-06-25): Fix #153 - Use C/N0 range on meter if SNR avg is above SNR max value
commit 3c29983b (2018-06-26): Fix #153 (for real) - Use C/N0 range on meter if avg SNR > max SNR value

This is a follow-on to d3f49c254f91c3557c40e34570a5f34865332e69 and a better solution, because:

1. It actually changes the position/color of the indicators and TextViews(!) - whoops, I missed this in d3f49c254f91c3557c40e34570a5f34865332e69.  That's what I get for trying to commit something quick while finishing my lunch.
2. It saves the state once the bad SNR value is detected so the C/N0 range is used for the duration of the Fragment.  This prevents the scale from oscillating back and forth when the SNR value is above [... cut]
closed 2018-06-26 by barbeau