barbeau/gpstest #157: C/N0 indicator tick marks aren't even

barbeau, 2018-06-19:
**Summary:** 

If you look at the C/N0 tick marks, some of them are strangely offset, while they should be evenly spaced.

[image]

**Steps to reproduce:** 

1. Tap on Sky in nav drawer
1. Look at Avg C/N0 scale

**Expected behavior:** 

Ticks below gradient should be evenly spaced

**Observed behavior:** 

Tick marks aren't evenly spaced

**Device and Android version:** 

Samsung Galaxy S8+ w/ Android 8.0

goldfndr, 2018-06-19:
It looks like there's a partial card (or something) on the screen at the extreme right of the screenshot. Look all the way to the right, starting at even with "28" and ending diagonally from the C/N0 legend.

I'm not seeing anything similar with either Dark or non-Dark Theme.

barbeau, 2018-06-19:
So that's actually a sliding side panel that's always available (i.e., the tab shows up on top of all apps) that Samsung has in their Galaxy S8 (and I'm guessing higher) Android implementations, so that's unrelated.

I believe this was caused either by me forgetting to update a few hard-coded values for the ticks when I changed the size of the meter previously, or some weirdness in layouts on different displays and Android versions.  I'm working on switching to a more dynamic implementation that would resize based on screen width, which should also help with https://github.com/barbeau/gpstest/issues/155.