barbeau/gpstest #161: Add more satellite shapes

barbeau, 2018-06-27:
**Summary:** 

I added originally added constellation shapes as more GNSS constellations came available in Android devices without a long-term strategy for how these would be managed.  As a result, the triangle is over-used, currently representing:
* Galileo
* QZSS
* All SBAS

We should refresh these to add new shapes to better distinguish between these constellations.

Current screenshot is below:

[image]

**Steps to reproduce:** 

From nav drawer tap on Sky view

**Expected behavior:** 

Have unique shapes per satellite constellation

**Observed behavior:** 

Triangle is used for:
* Galileo
* QZSS
* All SBAS

**Device and Android version:** 

N/A

goldfndr, 2018-06-28:
Some comments:

* Maybe use a square diamond (square rotated 45°) or lozenge (diamond with squeezed middle) for Galileo? If you want approximately the same area as GLONASS, extend height to about 1.4× (sqrt(2)), but perhaps GLONASS should be reduced in size to more closely match GPS's area. 
* Maybe upside-down triangle for QZSS? 
* IRNSS/NavIC will eventually happen.
* I haven't yet seen a screenshot with an SBAS satellite having elevation or azimuth. Given one's orbit, should these values be expected?

barbeau, 2018-06-28:
>Maybe use a square diamond (square rotated 45°) or lozenge (diamond with squeezed middle) for Galileo?

I think I'm going to use the triangle for Galileo, as it's easily distinguishable from other shapes and for many devices a number of Galileo devices will dominate the display.  QZSS has far fewer satellites and could be one of these other "stranger" shapes.

>IRNSS/NavIC will eventually happen.

Yes, thanks for bringing that up.  That's currently ticketed in https://github.com/barbeau/gpstest/issues/105, and I just added a note to the top post as well.

>I haven't yet seen a screenshot with an SBAS satellite having elevation or azimuth. Given one's orbit, should these values be expected?

Good question. Short answer is I don't know - I haven't seen an SBAS with elevation or azimuth yet either.  I'd welcome info from someone that might have more knowledge on that.

In the case of GPSTest, though, the Android GnssStatus API definitely allows SBAS satellite objects to have elevation and azimuth, so I have to design as if that could happen.

barbeau, 2018-07-10:
This site is useful for converting SVGs to Android vector XML:
http://inloop.github.io/svg2android/

commit 5ad3c50d (2018-07-10): Fix #161 - Add new shapes for QZSS, SBAS (#164)

New shapes:
* QZSS - Hexagon
* All SBAS - Diamond
* Add new constellation shapes to Legend and Help->About
closed 2018-07-10 by barbeau