barbeau/gpstest #168: Add simple benchmark using ground truth location

barbeau, 2018-07-11:
**Summary:** 

Analyzing the actual accuracy of GNSS in device has come up several times, especially in context of new dual frequency devices:
https://medium.com/@sjbarbeau/dual-frequency-gnss-on-android-devices-152b8826e1c

... like the Xiaomi Mi 8.  Users are referencing the "estimated horizontal accuracy" value from Location.getAccuracy():
https://developer.android.com/reference/android/location/Location.html#getAccuracy()

... as the actual accuracy, but it's not - it's the 68 percentile:

>We define horizontal accuracy as the radius of 68% confidence. In other words, if you draw a circle centered at this location's latitude and longitude, and with a radius equal to the accuracy, then there is a 68% probability that the true location is inside the circle.

I'd like to add simple benchmarking to GPSTest, which would include:
1. Allowing the user to enter a ground truth lat/long/alt (i.e., where they are actually located)
1. Display current horizontal, vertical, and estimated accuracy errors

I maintain another app, GPS Benchmark, that is better suited for in-depth analysis:
https://play.google.com/store/apps/details?id=com.gpsbenchmark.android

...so I'd like GPSTest just to focus on the current horizontal and vertical error, as well as the estimated accuracy uncertainty errors, and to display this info on the screen.  For simplicity GPSTest won't do more sophisticated things like recording error and calculating percentiles and outputting CSV and KML data, which GPS Benchmark currently does.

**Steps to reproduce:** 

1. Try to see the true (actual) accuracy of a position

**Expected behavior:** 

Give me a way to enter a ground truth value and see actual accuracy

**Observed behavior:** 

This isn't currently supported

barbeau, 2018-11-27:
Work-in-progress pull request is at https://github.com/barbeau/gpstest/pull/235.

bugeaud, 2018-12-30:
FYI Sean, GPS Benchmark requires people to use "menu button". But such a physical button is no more there since several Android iteration as explained :

https://android-developers.googleblog.com/2012/01/say-goodbye-to-menu-button.html

I've not found so far a way to access the required action on a Mi 8 for instance. I am afraid an update (even a small one) of the application is required to use it on modern phones.

barbeau, 2018-12-30:
@bugeaud In GPS Benchmark you should be able to access the virtual menu button if you hold down for 2-3sec the multitasking button on the left of the home button. Please give that a try!

commit fbc8faa1 (2019-04-05): Fix #168 - Add simple accuracy benchmarking (#235)

* Add material components
* Add bottom sliding panel
* Add graph views
* Accent colors change for buttons, checkboxes, etc.
* Bump dependencies
* Add "E." for estimated error label in Status
closed 2019-04-05 by barbeau