barbeau/gpstest #172: Chinese Languages

wangkf, 2018-07-19:
https://github.com/wangkf/gpstest/tree/master/GPSTest/src/main/res/arrays.xml
https://github.com/wangkf/gpstest/tree/master/GPSTest/src/main/res/strings.xml
Put these two files in the **gpstest-master\GPSTest\src\main\res\value-zh-cn** folder

barbeau, 2018-07-19:
Awesome, thanks @wangkf!  I believe the folder name should be `zh-rCN`?  This is the only way Android Studio would let me build the project, so I added it in https://github.com/barbeau/gpstest/commit/0b6c0ddd6b71e0a66b0c46917a229ff8e8e79326.

If you think this isn't right please let me know.  Also, if you'd be able to check the next beta release to make sure it looks right that would be great!

closed 2018-07-19 by barbeau

wangkf, 2018-07-20:
You are right.

commit 0b6c0ddd (2018-07-19): Fix #172 - Add Chinese translation