barbeau/gpstest #173: Update Chinese translation

barbeau, 2018-07-20:
@wangkf Would you be able to provide a translation for the new "What's New" text?

Commit with next text is https://github.com/barbeau/gpstest/commit/33df9870755987ed6df82408f735695033b8bee5.

Even if you just provide the translated text in this issue I can copy into the strings.xml file.

Thanks!

liuyunli, 2018-07-20:
"What's New"  text for Chinese is "本次更新内容"

barbeau, 2018-07-20:
Sorry, I wasn't clear. Here's the new text that needs to be translated:

~~~
Send feedback - Use this navigation drawer option to ask questions or report problems

New satellite shapes for QZSS and SBAS - See the updated shapes in the Sky view Legend

Chinese translation - Thanks to wangkf!
~~~

liuyunli, 2018-07-20:
Send feedback    发送反馈
Use this navigation drawer option to ask questions or report problems — 使用这个导航抽屉选项来询问问题或报告问题
New satellite shapes for QZSS and SBAS  为QZSS和SBAS卫星设计了新的图标
See the updated shapes in the Sky view Legend  ——请打开天空视图查看更新的图形
Chinese translation 中文翻译
Thanks to wangkf!  ，感谢 wangkf

barbeau, 2018-07-20:
Thanks @liuyunli!

commit 26e6ab4d (2018-07-20): Fix #173 - Update Chinese translation
closed 2018-07-20 by barbeau