barbeau/gpstest #175: Galileo satellites are shown with the wrong flag (UK instead of Europe)

amontefusco, 2018-07-24:
**Summary:** 

Galileo satellites are shown with the wrong flag.

**Steps to reproduce:** 

Just select STATUS panel.

**Expected behavior:** 

The right flag should be shown.

**Observed behavior:** 

The satellite ID 120 and ID 1243 are shown as a triangle in Sky view (so as Galuleo) but the flag is wrong (both in Sky view and Status panel)
 
**Device and Android version:** 

Samsung Galaxy A5 (2017) 
Model number: SM-A520F
Android version: 8.0.0
Samsung Experience version: 9.0
Google Play Service: version 12.8.74

**Screenshots:** 
[image]
[image]

goldfndr, 2018-07-24:
Keep scrolling in Sky and you'll see that triangles are used for more than Galileo. These two (120 and 123) are EGNOS SBAS satellites.

It appears to answer a question posed in #161. (Whether SBAS satellites ever have elevation and/or azimuth.)

amontefusco, 2018-07-24:
In the end, as I used another, almanac data based, PC based program (gpredict), in order to be sure that Galileo satellites are visible from my location, I should deduce that Android device is not able to receive Galileo.

Sound right?

closed 2018-07-24 by amontefusco

goldfndr, 2018-07-25:
This should be reopened, as satellite 123 is missing its flag (and shape).

It's good that you contributed the screenshot – see [#100 (comment)](https://github.com/barbeau/gpstest/issues/100#issuecomment-346363355).

barbeau, 2018-08-02:
Thanks for sharing the screensoht @amontefusco.  @goldfndr is right, in the current release triangles are used by more than Galileo, although this will change in the next release - see here for screenshots for the new shapes used for QZSS and SBAS:
https://github.com/barbeau/gpstest/pull/164

And yes, if you're in Europe and your device supports Galileo then it should be visible in GPSTest.  Here's a good place to see if your device supports Galileo - https://www.usegalileo.eu/EN/.

I opened a new issue for PRN/ID 123 without a flag/shape at https://github.com/barbeau/gpstest/issues/179.