barbeau/gpstest #176: Organize the head of the status tab in one column (with a bigger font), adding km/h

paulle, 2018-07-29:
The head of the status tab is organized in 2 columns (Lat, Long, Alt, Alt(MSL), Speed etc.
Maybe it would look better it it's organized in one column with a bigger font size for getting this information at a fast glance (especially for persons with eyes).
The second thing is to add the unit km/h for speed(I believe mainly European users would appreciate it), because km/h is common for measuring speed of cars, bikes and so on. One has a 'feeling' for this unit, with m/s or m/h one have recalculate, it's not so 'intuitive'.

barbeau, 2018-07-29:
@paulle Thanks for the feedback!  There is definitely a trade-off between information density and readability.  Given other user feedback I think the best solution is to have different options for layouts in the Status screen, some prioritizing density and others readability.

Having configurable units definitely makes sense too - most Americans would prefer miles/hour.

paulle, 2018-07-29:
Thank you for answer. Yes you are right, when users can choose between feasibility and density it's perfect, because younger people often prefer density and older ones prefer readability, because eyes aren't getting with age. And I believe too that mph is more 'intuitive' for Americans.

barbeau, 2018-09-11:
Also requested in https://github.com/barbeau/gpstest/issues/201 - have option for large speed display.

barbeau, 2018-09-14:
Configurable speed units have been requested a few more times, so I'm moving that to it's own issue at https://github.com/barbeau/gpstest/issues/207

Meteor0id, 2019-08-07:
Font size might be  made a user choice, but as of the current build the status card simply doesn't have enough room in some languages, forcing translators to use riddicules translations (just some letters) and forcing usrts to check to help menu just figure out what the abbriviation might mean even for simple words.

The status card should probably be height flexible, and the labels should be allowed to linebreak. Abbriviations should still be used, but they shouldn't be forced to be too short.

The same issue happens also at the bottom of the accuracy view, those labels also need linebreaks.

paulle, 2019-08-08:
Yes, making linebreaks and leaving the selection of the font size to the user would be a good solution.

barbeau, 2019-08-13:
The current Status view prioritizes information density over readability, and a lot of users really like this. So, for now I don't see major changes happening to the default Status view that would change this priority.

However, I'd like to add a secondary view of Status that would prioritize readability over density - maybe conceptually similar to this, where one or several data elements get their own cards:

[image]

Users would be able to toggle back and forth between these two views.

I'd welcome design suggestions for what this secondary view should look like.

Meteor0id, 2019-08-15:
I think I am advocating a different flavor of solution than @paulle is. I am advocating line-wraps: if the line get too long, only in that case take up another amount of height.
@paulle seems to advocate removing the two column layout which is not about linebreaks or linewraps, but rather about design.

I agree with Bareau that some amount of density is actually desirable, but I think it is too dens at this time and not flexible enough, so I advocate line-wraps if the text needs extra room.

paulle, 2019-08-15:
I believe I have to excuse me because of my poor English.
What I meant with linebreak
was only the option, when a line gets too long and get cutted and for this isn't completely readeable, in this case should be second line for seeing the whole text.
Same way as you put a text in a textfield and your text is longer than one line you finish your text in the second line.
I don't know really the difference between linebrek and linewrap. I hope I explained it now a little bit
better what I was meaning.

Meteor0id, 2019-08-16:
Maybe tap the status card to expand it into a more detailed overview (which also does not use abbreviations)?

closed 2021-10-01 by stale[bot]