barbeau/gpstest #177: Update German translation

barbeau, 2018-08-01:
**Summary:** 

@SIRSteiner Would you be able to translate the below string to German so I can include it the next release?

~~~
    <string name="main_help_whatsnew">&#8226;\u0020
        <b>Send feedback</b>
        - Use this navigation drawer option to ask questions or report problems\n&#8226;\u0020
        <b>New satellite shapes for QZSS and SBAS</b>
        - See the updated shapes in the Sky view Legend\n&#8226;\u0020
        <b>Chinese &amp; German translations</b>
        - Thanks to wangkf &amp; SIRSteiner!\n&#8226;\u0020
        <b>Bug fixes</b>
        - See http://bit.ly/gpstest-releases
    </string>
~~~

Even just commenting on this issue with the translation is fine - I can integrate from there.

Thanks in advance!

SIRSteiner, 2018-08-02:
I did a pull request ;-)

barbeau, 2018-08-03:
Fixed in https://github.com/barbeau/gpstest/pull/180. Thanks @SIRSteiner!

closed 2018-08-03 by barbeau