barbeau/gpstest #182: Update Chinese translation

barbeau, 2018-08-03:
**Summary:** 

@wangkf Sorry, one last request before this release - could you please translate the "What's new" text again?  This adds SBAS Astra 5B.  Thanks!

~~~
    <string name="main_help_whatsnew">&#8226;\u0020
        <b>Send feedback</b>
        - Use this navigation drawer option to ask questions or report problems\n&#8226;\u0020
        <b>New satellite shapes for QZSS and SBAS</b>
        - See the updated shapes in the Sky view Legend\n&#8226;\u0020
        <b>Support for SBAS Astra 5B</b>
        - Now visible in Status and Sky views\n&#8226;\u0020
        <b>Chinese &amp; German translations</b>
        - Thanks to wangkf &amp; SIRSteiner!\n&#8226;\u0020
        <b>Bug fixes</b>
        - See http://bit.ly/gpstest-releases
    </string>
~~~

liuyunli, 2018-08-04:
<b>Send feedback</b>
<b>发送反馈</b>
        - Use this navigation drawer option to ask questions or report problems
        - 使用这个导航抽屉选项来询问问题或反馈错误\n&#8226;\u0020
<b>New satellite shapes for QZSS and SBA <b>
<b>为QZSS和SBAS卫星设计了新的图标</b>
    - See the updated shapes in the Sky view Legend\n&#8226;\u0020   
     - 请打开天空视图查看更新的图形\n&#8226;\u0020
        <b>Support for SBAS Astra 5B</b>
        <b>支持 SBAS Astra 5B<.b>
        - Now visible in Status and Sky views\n&#8226;\u0020
       - 在状态试图和天空试图可以看到
        <b>Chinese &amp; German translations</b>        
         <b>中文翻译,德文翻译</b>
       - Thanks to wangkf &amp; SIRSteiner!\n&#8226;\u0020
        - 感谢 wangkf , SIRSteiner！ \n&#8226;\u0020
 <b>Bug fixes</b>      
  <b>错误修复</b>
- See http://bit.ly/gpstest-releases
- 点击 http://bit.ly/gpstest-releases

barbeau, 2018-08-07:
Thanks @liuyunli!

commit 51500bfb (2018-08-07): Fix #182 - Update Chinese translation
closed 2018-08-07 by barbeau