barbeau/gpstest #190: Create table for device supporting various features

barbeau, 2018-08-11:
**Summary:** 

The "Send feedback" feature includes device model numbers and whether they support various features in GPSTest such as raw measurements, nav messages, NMEA, injecting extra data, etc.

I'd like to create a table with this info.  I'm going to log them in this issue until I create the table so I don't lose them.

barbeau, 2018-08-11:
App version: v3.1.3 (32-osmdroid)
Model: Redmi 4A
Android version: 7.1.2 / 25
GNSS HW Year: 2015
Raw measurements: NOT_SUPPORTED
Navigation messages: NOT_SUPPORTED
NMEA: SUPPORTED
Inject XTRA: NOT_SUPPORTED

barbeau, 2018-08-11:
App version: v3.1.3 (32-osmdroid)
Model: HTC One M9
Android version: 7.1.2 / 25
GNSS HW Year: 2015
NMEA: SUPPORTED

JonnyTech, 2018-08-11:
Wileyfox Spark+ (porridge):

```
App version: v3.1.3 (32-osmdroid)
Model: Spark
Android version: 7.1.2 / 25
Location: gps <removed>
GNSS HW Year: 2015 or older 
NMEA: SUPPORTED
```

barbeau, 2018-08-11:
@JonnyTech Thanks!  By default the email only contains the features that the user has enabled/tried.  If you want to provide a full record, in the nav drawer you can try:
* Inject xtra
* Inject time
* Clear assist (caution - on some devices it can take a while to recover from this.  I'd confirm you can inject the above before trying this)

...and in Settings, you can check logging:
* GNSS measurement data
* Nav messages

If you send a "Send feedback" email after doing this, it will capture success/failure for all the above.

JonnyTech, 2018-08-11:
Done. Original post updated and feedback sent.

barbeau, 2018-08-11:
@JonnyTech Great, thanks!

goldfndr, 2018-08-12:
App version: v3.1.4 (33-google)
Model: XT1028
Android version: 4.4.4 / 19
Location: gps (removed), 10 second(s) ago
nullNMEA: SUPPORTED
Inject XTRA: SUPPORTED
Inject time: SUPPORTED
Delete assist: SUPPORTED

Google Play Services App: 12.8.74 (000306-204998136)
Google Play Services Library: 11910000

I'm guessing the "null" before "NMEA" is due to my pre-7.0 Android version.

barbeau, 2018-08-13:
App version: v3.1.3 (32-osmdroid)
Model: H8266
Android version: 8.0.0 / 26
Location: gps , 0 second(s) ago
GNSS HW Year: 2017
NMEA: SUPPORTED
Inject XTRA: NOT_SUPPORTED
Inject time: SUPPORTED
Delete assist: SUPPORTED

commit f94dc762 (2018-08-14): Fix #190 - Add table with device support info
closed 2018-08-14 by barbeau