barbeau/gpstest #202: Xiaomi Mi 8 has invalid Beidou carrier frequency

barbeau, 2018-09-10:
**Summary:** 

Reported via user screenshot here:
https://medium.com/@trungthanh1288/this-is-mi-8-with-newest-android-p-in-beta-testing-program-655f95a5e376

The existing floating point value used for the Beidou carrier frequency of `1575.450` for Beidou doesn’t seem to match [any known Beidou values](https://en.wikipedia.org/wiki/BeiDou_Navigation_Satellite_System#Frequencies).

Is this a bug in the Mi 8 Android P beta, or a valid Beidou carrier frequency?

Here is a full mapping of our current Beidou carrier frequencies in code is below, from `GpsTestUtil`:

~~~
                // GnssType.BEIDOU
                if (MathUtils.fuzzyEquals(carrierFrequencyMhz, 1561.098f, TOLERANCE_MHZ)) {
                    return "B1";
                } else if (MathUtils.fuzzyEquals(carrierFrequencyMhz, 1589.742f, TOLERANCE_MHZ)) {
                    return "B1-2";
                } else if (MathUtils.fuzzyEquals(carrierFrequencyMhz, 1207.14f, TOLERANCE_MHZ)) {
                    return "B2";
                } else if (MathUtils.fuzzyEquals(carrierFrequencyMhz, 1176.45f, TOLERANCE_MHZ)) {
                    return "B2a";
                } else if (MathUtils.fuzzyEquals(carrierFrequencyMhz, 1268.52f, TOLERANCE_MHZ)) {
                    return "B3";
                }
~~~

**Steps to reproduce:** 

Run GPSTest on Xiaomi Mi 8 Android P beta in a location where you can see Beidou satellites

**Expected behavior:** 

GPSTest should show one of the following labels if the given carrier frequency matches a known Beidou carrier frequency:
* B1
* B1-2
* B2
* B2a
* B3

**Observed behavior:** 

The raw carrier frequency of `1575.450` is shown in the app

**Device and Android version:** 

Xiaomi Mi 8 Android P beta

barbeau, 2018-11-14:
As quoted in https://medium.com/p/152b8826e1c/info:

>Beidou is now in its 3rd phase of development (BDS-3), the BDS space constellation will gradually transition from BDS-2’s to BDS-3’s.
>
>The basic space constellation of BDS-3 consists of 3 GEO satellites, 3 IGSO satellites, and 24 MEO satellites.
>
>BDS-3 defines three open service signals:
>
>* B1C: 1575.42 MHz with a bandwidth of 32.736 MHz
>* B2a: 1176.45 MHz with a bandwidth of 20.46 MHz
>* B3I: 1268.52 MHz with a bandwidth of 20.46 MHz”

Looks like the signal we're seeing on the Mi 8 is the B1C, so I can add a label for that.

goldfndr, 2018-11-15:
I'm getting a 403 error result code when going to https://medium.com/p/152b8826e1c/info -- is that a private conversation? That's even after logging in to Medium. OTOH, I see the quote at https://www.everythingrf.com/community/beidou-frequency-bands.

But here's another table of frequencies:

* B1 1575.420 MHz
* B2 1191.795 MHz
* B3 1268.520 MHz
* Bs 2492.028 MHz

from [this PowerPoint PDF](http://www.unoosa.org/documents/pdf/icg/2017/05_icg12.pdf) of 2017-12 on page 6. It's quite something that B1 and B2 each have two frequencies, eh? :)

barbeau, 2018-11-15:
Try https://link.medium.com/fYmf5lmpSR instead, that works for me in incognito mode.

commit e0a6e4a8 (2018-11-21): Fix #202 - Add B1C Beidou carrier frequency
closed 2018-11-21 by barbeau