barbeau/gpstest #203: Xiaomi Mi 8 shows all satellites being used in a fix

barbeau, 2018-09-10:
**Summary:** 

Reported via user screenshot here:
https://medium.com/@trungthanh1288/this-is-mi-8-with-newest-android-p-in-beta-testing-program-655f95a5e376

It’s a little suspicious that the device is reporting using ALL satellites (“U” indicator in “Flags” column) to compute a fix.

Is this a bug in the Mi 8 Android P beta, or a valid use of all satellites in a fix?

**Steps to reproduce:** 

Run GPSTest on Xiaomi Mi 8 Android P beta in a location where you can see Beidou satellites

**Expected behavior:** 

Most likely all satellites aren't used to compute a fix - some should be missing a "U" flag

**Observed behavior:** 

For all satellites, the "Flags" column has a "U"

**Device and Android version:** 

Xiaomi Mi 8 Android P beta

barbeau, 2018-09-10:
I'm closing this as "Won't fix" as it's almost certainly a bug in Xiaomi Mi 8 (or expected behavior), but I wanted to document this somewhere.

closed 2018-09-10 by barbeau

goldfndr, 2018-09-11:
I'm able to find [a couple](https://imgur.com/gallery/IfoNmNV) [of screenshots](https://china-gadgets.com/xiaomi-mi-8-smartphone-review/) that show fewer than *all* satellites in use. Most screenshots *do* show all satellites in use.