barbeau/gpstest #208: map view within GPStest is only showing a white space

paulle, 2018-09-14:
The map view within GPStest is only showing a white space (some times with pin-point, sometimes without it).
sony xz2, android8.0, GPStest 3.20 from f-droid
[image]
[image]

paulle, 2018-09-15:
[image]
That's the map showing less than a second if I switch to map view.

paulle, 2018-09-18:
@barbeau Thank you for your answer, and yes i n my case, if I pan around it loads slowly too, sometimes complete sometimes the map isn't complete.
My personal workaround for the moment is, pan around and waiting a little bit, or in other cases I use OSMand for map view (and navigation). OSMand has the advantage that I can have stored a lot of maps on the sd-card of my phone and I don't need data-connection or wlan (for me a great advantage when travelling).
So for me it's ok if you stay with the default OSM network tile server, because I know now that I have to pan around and wait a little bit for seeing the map. Maybe it could be useful adding a hint that one has to pan around and wait a little bit because of slow loading for avoiding that people think map view don't work (like me).
But I don't know what other users think about this point.

barbeau, 2019-04-19:
This is fixed as part of https://github.com/barbeau/gpstest/pull/235 - Mapbox is the current tile provider for the OSMDroid version.

closed 2019-04-19 by barbeau