barbeau/gpstest #210: Galileo

PetrBrno, 2018-09-15:
**Summary:** 

Unstable depiction of galileo satelites.

**Steps to reproduce:** 

In STATUS view. Satelite's depiction.

**Expected behavior:** 

All GNSS would behave in a similar way. 
Appears and disappears one by one (concrete satelite), not all satelites of one GNSS at once. 
Simply as it must physically be. 

**Observed behavior:** 

GPS, Glonass are stable depicted (every time you can see some of these satelites). In the contrary Beidou and Galileo shows just sometimes and for a few seconds. It's easy to notice that all satelites of one sort (Galileo, Beidou) disappears at once and at once appears again. For example six satelites in one step, and every time all of one kind together, but never GPS and Glonass. It's surely not real physical state...

**Device and Android version:** 

Sony Xperia XZ Premium G8141, Android 8, It is the stock version from the manufacturer, APP  v3.1.5(34) GNSS HW Year 2016.

**Screenshots:** 

[image]

goldfndr, 2018-09-15:
For whatever it's worth, I see similar behavior on my Moto G (2013). The Navstar satellites are pretty much always there, the GLONASS appear and disappear as a group.

Observed in other apps (the counts (used/seen) will precipitously rise or drop).

PetrBrno, 2018-09-16:
This problem is probably connected with android. There would be some bug. The system don't deal with all GNSS equally.
As goldfndr showed other devices has the same problem. The same problem at different producers?

barbeau, 2018-09-17:
Different device OEMs can use the same chipset and underlying native code that generates the GNSS updates which are passed to the Android platform, so most likely the problem lies there.