barbeau/gpstest #218: Broken on Android P Version 9.0

RavetcoFX, 2018-09-27:
**Summary:** 

On the Google pixel xl 1 the status seems to be completely broken as it doesn't show any information apart from some of the SATs. I'm on Android 9.0 Sep 5 security patch, and the app has location permission

**Steps to reproduce:** 

Run the app in Android 9.0

**Screenshots:**

barbeau, 2018-09-27:
@RavetcoFX strange. Could you please provide a screenshot? Did you try power cycling your device in a location with good Internet access and a clear view of the sky? You could also try the "Inject XTRA" and "Inject time" options.

RavetcoFX, 2018-09-28:
Hi @barbeau 
I get "Device does not support injecting XTRA data" and injecting time data seemingly does nothing. There is no difference when getting a clear view of the sky, and gps works as normal in Google Maps, snapchat etc.
[image]

barbeau, 2018-09-29:
Thanks for the info and screenshot! 

A few more questions:
* Are you using the latest GPSTest beta? https://github.com/barbeau/gpstest/blob/master/BETA_TESTING.md
* Did you try power cycling your device? Sometimes this fixes issues with getting a GPS fix 
* Did you use the "Clear assists data" option? I've seen this cause issues on some devices like you're seeing.

Other apps use the "fused" Location provider, which can also use Wi-Fi and cell towers (among other things) to get a location. GPSTest uses only GPS, and it appears your device can't get a pure GPS fix.

RavetcoFX, 2018-10-05:
Beta worked. Closing issue

closed 2018-10-05 by RavetcoFX

barbeau, 2018-10-05:
Thanks for letting me know!