barbeau/gpstest #220: Add PRN131 for WAAS

ndoggac, 2018-10-04:
WAAS GEO5 (AKA PRN 131) recently came online, replacing PRN 133 which went offline several months prior.  However, PRN 133 will be reused (different space vehicle) when GEO6 comes online next year, and PRN 135 will be turned off several weeks after 133 comes back online.

[GEO5 Reference](https://space.skyrocket.de/doc_sdat/eutelsat-117-west-b.htm)
[GEO6 Reference](https://space.skyrocket.de/doc_sdat/ses-15.htm)

The GEO7 contract was awarded to Leidos, and is in development.
[GEO7 Announcement](https://investors.leidos.com/news-and-events/news-releases/press-release-details/2018/Leidos-Awarded-GEO-7-Task-Order-to-Enhance-US-Air-Traffic-System/default.aspx)

[PRN Code Assignments](https://www.losangeles.af.mil/About-Us/Fact-Sheets/Article/343695/pseudorandom-noise-code-assignments/)

barbeau, 2018-10-04:
Thanks for the info @ndoggac!  I just overhauled the SBAS presentation in https://github.com/barbeau/gpstest/pull/217, which now shows satellites as their constellation, not their operator.  Here's the code for SBAS:

~~~
    public static SbasType getSbasConstellationType(int svid) {
        if (svid == 120 || svid == 123 || svid == 126 || svid == 136) {
            return SbasType.EGNOS;
        } else if (svid == 131 || svid == 133 || svid == 135 || svid == 138) {
            return SbasType.WAAS;
        } else if (svid == 127 || svid == 128 || svid == 139) {
            return SbasType.GAGAN;
        } else if (svid == 129 || svid == 137) {
            return SbasType.MSAS;
        }
        return SbasType.UNKNOWN;
    }
~~~

So this will assign the WAAS flag (USA) to PRNs 131, 133, 135, and 138 no matter what satellite is broadcasting them.

I'll leave this issue open, though, so I can update the satellite name assignment.  Satellite names aren't being surfaced in the current GPSTest UI in the master branch but I'll likely add them as satellite details in the future.

ndoggac, 2018-10-05:
Thanks for the lightning response!! 

Maybe you have all these links already, but if you ever wanted to add further details regarding GPS, Glonass, and Galileo satellites, you can parse the following tables.

[Coast Guard GPS Constellation Status](https://www.navcen.uscg.gov/?Do=constellationstatus)
II-A are the oldest SV type in service, followed by II-R, II-RM (added L2C), and II-F (added L5).  The first GPS III satellite (adds L1C, etc) may be launched by end of 2018 (total 10), followed by III-F satellites in the future.  

[Glonass Website's Status for GPS](https://www.glonass-iac.ru/GPS/)
Provides the launch and "in-service" dates. 

[Glonass Website Status for Glonass Constellation](https://www.glonass-iac.ru/GLONASS/)

[Galileo Constellation Status](https://www.gsc-europa.eu/system-status/Constellation-Information)

barbeau, 2018-10-06:
@ndoggac Thanks!  That definitely helps.  Do you know of a site that maps the actual satellite name to the broadcasting PRN?  That seems to be the main thing I'm missing.  Ideally I'd like users to be able to tap on a row in the Status view and pop up info about that satellite with a link to the web for more info.

ndoggac, 2018-10-06:
[Wikipedia - List of GPS satellites](https://en.wikipedia.org/wiki/List_of_GPS_satellites)

[Wikipedia - List of Galileo satellites](https://en.wikipedia.org/wiki/List_of_Galileo_satellites)

[Wikipedia - List of Glonass satellites](https://en.wikipedia.org/wiki/List_of_GLONASS_satellites)

[Wikipedia - List of Beidou satellites](https://en.wikipedia.org/wiki/List_of_BeiDou_satellites)

[N2YO](https://www.n2yo.com/satellites/?c=20)
Go to "Satellites On Orbit" menu, and choose desired system under "Categories".  Tons of additional info per satellite that can be parsed.

[QZSS Website - List of positioning satellites](http://qzss.go.jp/en/technical/satellites/index.html)

barbeau, 2018-10-29:
Here is a link that defines the PRN assignments maintained by the U.S. Air Force, which seems to be the best reference for SBAS PRN-to-satellite-name mappings:
https://www.losangeles.af.mil/About-Us/Fact-Sheets/Article/343695/pseudorandom-noise-code-assignments/

ndoggac, 2018-10-30:
Agreed, it was linked in my first post. ;-)

I'm not too familiar with the Android logs (API) you are utilizing for data in your display.  Is there a flag that denotes SBAS corrections are provided for a given GPS satellite?  If so, could that be integrated into the display?  Also, I noticed there is not elevation or azimuth values for SBAS satellites.  Is that information provided in the logs?

barbeau, 2018-10-30:
@ndoggac Ah, yes, thanks!  :)  Forgot about that.

Here's the GnssStatus object that is provided by the Android platform for each satellite, which GPSTest shows in Status view:
https://developer.android.com/reference/android/location/GnssStatus

The `getConstellationType()` method says whether the satellite is SBAS or one of the GNSS types (see top of page in link above).  I have to break down the SBAS into individual constellations based on the satellite IDs.

>Is there a flag that denotes SBAS corrections are provided for a given GPS satellite? 

Unfortunately no

>Also, I noticed there is not elevation or azimuth values for SBAS satellites. Is that information provided in the logs?

I've noticed this on my device too (Samsung Galaxy S8+).  Some devices do provide this info as I've seen it in other screenshots.  Elevation and azimuth come from the same `GnssStatus` object linked to above.  If the values are blank in the UI, that means that the `GnssStatus` object didn't provide values for that satellite.  I don't know why SBAS satellites typically don't have this info, unless maybe the device is decoding the correction info for other satellites from that SBAS satellite but not the L1 broadcast.  Only the device OEM or GNSS chipset manufacturer could answer that.

commit 6723178f (2019-03-19): Fix #220 - Add satellite name for GEO5 (currently not shown in UI)
closed 2019-03-19 by barbeau