barbeau/gpstest #233: Status view sorts by the wrong columns

kevpr, 2018-10-26:
**Summary:** 

In the status view, some sort orders sorts by other criteria/columns than the one(s) chosen.

**Steps to reproduce:** 

- Open GPSTest
- choose status view
- sort by signal strength

**Expected behavior:** 

List should by sorted by contents of the SNR column. (preferrably descending)

**Observed behavior:** 

List is sorted by ID column ascending, regardless of SNR.

Some other sort options are similarly wrong, or somewhat strange. I use Android, and this app, in German. So this might well be a translation issue. My sort-by menu offers: (added in my words, what that means in English)
1. Konstellation (constellation)
2. Signalstärke (signal strength)
3. Trägerfrequenz (carrier frequency)
4. Zum Fix verwendet (used for fix)
5. Konstellation, Trägerfrequenz (see 1, 3)
6. Konstellation, Signalstärke (see 1, 2)
7. Konstellation, zum Fix verwendet (see 1, 4)

The column(s) it seems to sort by:
1. ✅ ID ascending (this sort order is valid as GNSS and as carrier frequency too, on my devices)
2. ✖️ same as 1.
3. ✖️ SNR descending
4. ✅ Flags "U" (first "U", then no "U"), then ID ascending
5. ✅ same as 1.
6. ? GNSS, then SNR ascending (Descending would make much more sense to me!)
7. ? GNSS, then Flags "U" (first no "U", then "U"; the other order would make much more sense to me!)

✅ = Expected behavior, afaict with my limited devices.
✖️ = This is different behavior than I expect.
? = This behavior I find not completely wrong, but improvable.

I've observed this in GPStest 3.2.5 and 3.2.4 from f-droid, on two seperate devices. Didn't check other versions.

**Device and Android version:** 

1. Motorola Moto G 2nd Gen with 4G LTE (XT1072 "THEA")
    1. (with NAVSTAR GPS & GNSS, single frequency, no SBAS)
    2. Android 6.0 stock Motorola, not rooted
2. Samsung Galaxy Nexus ("MAGURO")
    1. (with NAVSTAR GPS only, single frequency, no SBAS)
    2. Android 6.0.1 LineageOS 13.0-20180211-NIGHTLY-maguro stock build, rooted

**Screenshots:** 

Upon request. If so, please let me know if the XT1072's screenshots would be enough, (my other phone behaves the same, but has only one GNSS) and whether you need me to switch languages first, but we might be able to skip that, see below.

**Prospect**

I'd be glad if someone could first quickly check this on other language phones than German, in order to tell whether or not this might really be a translation error. If so, I'd gladly contribute correct translations, if you tell me what the original is. Please let me know if I can offer any other support in fixing this issue, perhaps testing any other APKs.

best regards
Kevin Price

barbeau, 2018-10-29:
@kevpr Thanks for opening this issue!

Here's what the text looks like in English:

[image]

### Re: issues 2 and 3

Looking at the German translation for the lists:
https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/res/values-de-rDE/arrays.xml

It looks like the order of the `sort_sats` is wrong and doesn't match the English order in:
https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/res/values/arrays.xml

In other words, entry 2 and 3 are swapped.  I'll go ahead and fix this, but please let me know if you think differently.

### Re: 6 and 7

This is an issue with how sorting is implemented on Android API level 23 (Android 6.x M and lower).  These Android versions don't have great support for sorting by more than one attribute, so the order isn't ideal.

barbeau, 2018-10-29:
@kevpr Also, if you'd like to contribute to improving translations, I just added some documentation here for this:
https://github.com/barbeau/gpstest/blob/master/TRANSLATIONS.md

Any help is welcome!

closed 2018-10-29 by barbeau

kevpr, 2018-10-29:
Dear Sean!

Thanks a lot for your screenshot, and for your thoughts. That clarified, as I suspected, that this was in fact a mere translation bug, 'sort_stats' 2 and 3 having been swapped. Renaming this issue, and swapping back the strings seems good to me for now.

Re: 6 and 7: What a pity, but so be it. :-| (Somehow: Damn. Sorting algorithms are no rocket science, and particularly sort orders are not. This is still bugging me, but it's no matter of this issue, it's just a matter of representation.)

I'll check your translation doc. Thanks for pointing out. I'd be glad to help. Maybe just in proofreading.

ATM I cannot afford to buy a Xiamoi Mi 8, but I'm very excited about it. I do understand hyperbolic navigation, and GNSS theory to some extent, and I'm well aware of the gpsd and the GNSS-SDR projects.

I'd really love to build some receiver hardware and logic on my own from scratch. Perhaps eventually on a RISC-based FPGA with hardware DSP. ("Can we fix it?" ;D ) Including enhancements such as network-based approximate time and positions, almanac/epheremis, and barometric sensor data, contributing to a quick fix, and ionospheric downloads and SBAS/WAAS/D-GPS, as well as phase-measurements for some real kick-ass precision.

Regardless of any of that: Thanks for resolving this isssue!

Best
Kevin

barbeau, 2018-10-29:
>Re: 6 and 7: What a pity, but so be it. :-| (Somehow: Damn. Sorting algorithms are no rocket science, and particularly sort orders are not. This is still bugging me, but it's no matter of this issue, it's just a matter of representation.)

It bugs me too :). If someone wants to work on this for older Android versions (i.e., implementing custom Comparators), here's the current implementation:
https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/java/com/android/gpstest/util/SortUtil.kt#L21

barbeau, 2018-10-30:
@kevpr I just released v3.2.6 to Google Play and it should be available to download shortly.  Please take a look and let me know if the translation issue looks like it's fixed.

kevpr, 2018-11-06:
Sry I saw this just now. Thank you for the quick fix! (brilliant TTFF!) Yes, 3.2.6 (which I got from f-droid, rather than from Google Play) clearly resolves the core issue on all my devices. That other thing about the sort order is obviously not sorted out, but for now I cannot come up with any good ideas on that. (Don't blame me for the puns. :) )

commit 7236f283 (2018-10-29): Fix #233 - Swap order of sort labels in German to match English