barbeau/gpstest #236: GPS Leap Seconds Not Utilized in Time Calculation

ndoggac, 2018-11-02:
Totally nit-picking here, but thought I would let you know in case you wanted to learn about it.  The time calculation in the app is not properly utilizing the GPS leap second in its calculation.  You can recognize this by watching your cell phone time change over from one minute to the next while simultaneously watching the app's time.  Cell phone time, provided by GPS and properly adjusted by cellular stations via the data network, will change over when the app's second hits 18 seconds after the minute.

The current GPS leap second value is 18, and it is announced when it should be adjusted.  You can read way more than you probably want to know about it [here](https://tycho.usno.navy.mil/leapsec.html).

You can usually find announcements or pertinent news on the subject on [GPSWorld](http://gpsworld.com/tag/leap-second) or [InsideGNSS](http://insidegnss.com/?s=leap+second) websites.  Looks like the current leap second value is good until at least 2023.

barbeau, 2018-11-03:
@ndoggac Thanks for sharing!  That's interesting, I've never actually stared at this before.

Looks like this is device-specific, though, if I'm understanding you right.  The time shown in GPSTest is a straight conversion from Android `Location.getTime()` in UTC time:
https://developer.android.com/reference/android/location/Location.html#getTime()

...and this value comes from the GPS hardware.

For example, if I'm understanding you right, my Samsung Galaxy S8+ on Sprint in the U.S. (SM-G955U) is taking the leap second into account because the GPS time shown in GPSTest in turning over to the next minute in sync with the system clock in the upper right:

[image]

I'm curious what device you're running and on what network.

closed 2018-11-03 by barbeau

ndoggac, 2018-11-03:
Ok, wow....very interesting.  My cellular/android time at the very top right does not increment until the time in your app reaches 18 seconds.  I think you're right, must be hardware dependent, because...

My Device that exhibits the 18s problem:
Moto X4 (XT1900-1 64GB Storage/4GB RAM with Snapdragon 630 chipset) on Project Fi (Sprint/Tmobile).

Another device that does not exhibit the same issue, rolls over at same time:
Moto G6 (32GB Storage/3GB RAM wtih Snapdragon 430 chipset) on Project Fi (Spring/TMobile)

ValZapod, 2018-12-02:
I just leave it there http://www.leapsecond.com/java/gpsclock.htm 
"the GPS time shown in GPSTest" there is no GPS time in your app, it's UTC time. Just because  Location.getTime() is already in UTC, as written in Android docs.