barbeau/gpstest #238: no CF on Xiaomi dual frequencies cell phones

benzfish, 2018-12-04:
**Summary:** 

Summarize your issue in one sentence (what goes wrong, what did you expect to happen)

**Steps to reproduce:** 

How can we reproduce the issue?

**Expected behavior:** 

What did you expect the app to do?

**Observed behavior:** 

What did you see instead?  Describe your issue in detail here.

**Device and Android version:** 

What make and model device (e.g., Samsung Galaxy S3) did you encounter this on?  What Android version (e.g., Android 4.0 Ice Cream Sandwich) are you running?  Is it the stock version from the manufacturer or a custom ROM?  What version of Google Play Services is running on your device?

**Screenshots:** 

Can be created by pressing the Volume Down and Power Button at the same time on Android 4.0 and higher.

benzfish, 2018-12-04:
Summary:

no CF shows in Xiaomi8 and MIX3. Both of them are dual frequencies phones

Steps to reproduce:

open app, and see the CF column.

Expected behavior:

GPS/GLONASS/Galileo/BDS/QZSS should show CF

Observed behavior:

nothing here on CF column, version is 3.2.7

Device and Android version:

Xiaomi MIX3 dual frequencies phone with Android P.
This issue only happens at version 3.2.7, but works well at version 3.2.6

Screenshots:

N/A

brucechase, 2018-12-06:
I have a Mi8.  Before I did an update to the operating system, the phone provide no CF values.  After I went through the hassle of upgrading to Android Pie (version 9) Beta, I found that the CF values are now filled in.  I believe that Xiaomi will be making the Android Pie version available for automatic updates in the next month or so when it is off Beta version.

benzfish, 2018-12-14:
@barbeau sorry. i must make a mistake. i tried both of the apks with different version. both of them could show CF now. sorry.

closed 2018-12-14 by benzfish