barbeau/gpstest #242: Add degree/minutes/decimals (standard Geocaching) format

mirabilos, 2018-12-18:
**Summary:** 

Please add DDD°MM.MMM in addition to DDD.D…° and DDD°MM′SS.SS″ formats

**Device and Android version:** 

HTC Desire, Android 4.4 (LiquidSmooth 3.2), MicroG

barbeau, 2018-12-19:
Thanks for the feedback @mirabilos!

@firebovine Any interest working on this? :)

firebovine, 2018-12-19:
Hmm. With multiple formats we'll probably want a select menu similar to "Choose Map Type". I will look at LOE this weekend if I have time.

Any chance there are other representations we should add?

mirabilos, 2018-12-19:
firebovine dixit:

>Hmm. With multiple formats we'll probably want a select menu similar
>to "Choose Map Type". I will look at LOE this weekend if I have time.

Thanks!

>Any chance there are other representations we should add?

The three I named plus UTM in the format

	32U 461344 5481745

that is, zone, quadrat (which implicitly contains north/south),
a space, then the “right value” (longitudinal part), a space,
then the “up value” (latitudinal part) are common.

I don’t know of any other globally common formats, and local
formats like Gauß-Krüger or British Ordnance are being replaced
by UTM (at least in civilian use, authorities and emergency
services are following slowly though AFAIK).

All four are usually referenced in WGS84.

barbeau, 2018-12-19:
@mirabilos What is UTM commonly used for in context of GPSTest?  Is it commonly used for geocaching too?

>With multiple formats we'll probably want a select menu similar to "Choose Map Type"

@firebovine Yes, I agree.

>Any chance there are other representations we should add?

@firebovine Not that I know of (at least that would be commonly used with GPSTest).

mirabilos, 2018-12-19:
Sean Barbeau dixit:

>@mirabilos What is UTM commonly used for in context of GPSTest?  Is it
>@commonly used for geocaching too?

GPSTest shows the current coordinates. It is often useful to have them
in UTM format, especially when talking to authorities or emergency ser‐
vices. It is also used in Geocaching, although rarely; that has largely
standardised on DDD°MM.MMM format.

goldfndr, 2018-12-20:
There's https://github.com/mvglasow/satstat/issues/127 -- a request for Maidenhead by @tomikmar in a similar app. The format is pretty trivially extended from six to eight characters (quarter minutes or 15'' for longitude, eighth minutes or 7.5'' for latitude), but beyond there's less consensus on what to use.

mirabilos, 2018-12-20:
Richard Finegold dixit:

>There's https://github.com/mvglasow/satstat/issues/127 -- a request
>for Maidenhead by @tomikmar in a similar app. The format is pretty

Oh, interesting. I wrote a shell script to convert from Maidenhead
for a curiosity once, but had not thought it in widespread use,
considering it was the key to solving a mystery. See:

https://evolvis.org/plugins/scmgit/cgi-bin/gitweb.cgi?p=shellsnippets/shellsnippets.git;a=blob;f=mksh/maidenhead;hb=HEAD

firebovine, 2018-12-21:
Support for Maidenhead and/or UTM could be added once #243 is merged, but they both require a bit of code/other libraries as they are a bit more than just reformatting. I reached out to Chris to see if we could use his code for Maidenhead. I saw a UTM library, but I didn't look into its license yet. I'll let @barbeau decide if he wants them added included before pursuing further.

mirabilos, 2018-12-23:
> beyond there's less consensus on what to use.

I’ve encountered up to twelve characters during my research, just going on in alternating twos.

Shell extglob (regexp) for that:

    [A-R][A-R]?([0-9][0-9]?([A-X][A-X]?([0-9][0-9]?([A-X][A-X]?([0-9][0-9])))))

In PCRE this would be:

    /^[A-R][A-R](?:[0-9][0-9](?:[A-X][A-X](?:[0-9][0-9](?:[A-X][A-X](?:[0-9][0-9])?)?)?)?)?$/

What’s that library you saw? I’m used to doing licence analysēs, when packaging for BSD ports, and as a Debian Developer.

barbeau, 2018-12-23:
@firebovine Thanks for working on this!  Quick reply here (I haven't had a chance to review PR code yet) - I'm fine merging DMM support without Maidenhead or UTM, since to my knowledge DMM is much more widely used.  That being said I'm certainly not against adding Maidenhead or UTM if someone wants to go to the effort of implementing it.  If we're using code from somewhere else it would need to be compatible with Apache v2.0 (ideally re-licensed under Apache) and cited properly.

mirabilos, 2018-12-25:
I’d **personally** be happy with just the location, that is

    N 50 42.000 E 007 06.000

GPS Status also includes a map link. An example which we could use would be:

    I am here: https://www.openstreetmap.org/?mlat=50.7&mlon=7.1 (N 50 42.000 E 007 06.000)

Omitting the degree sign `°` makes this pure ASCII, which may be fortunate for some users.

The OSM link accepts marker latitude and longitude as decimals (with an optional `-` for minus before) with up to seven decimals, i.e. something like `/?mlat=49.1&mlon=-123.1234567` for somewhere in America. (GPS Status uses a Google Maps link, but then, it’s a Play Store application… I’d prefer not to.)

barbeau, 2018-12-26:
@mirabilos have you tried the "Share" icon that pops up in the top action bar after you get a fix? It will give you a lot of options to share a URL formatted for GeoHack, which is a meta-formatter for a lot of different services (including OSM AND Google Maps):

https://tools.wmflabs.org/geohack/geohack.php?params=29.72629576;-81.9470374

One of the share options is also "Copy to clipboard". 

Please take a look and let me know if that meets your needs.

mirabilos, 2018-12-27:
Sean Barbeau dixit:

>@mirabilos have you tried the "Share" icon that pops up in the top

Hm, I don't have a "Share" menu entry, only a "Send Location"...

>One of the share options is also "Copy to clipboard".

... which does not include this.

Perhaps I missed a recent upgrade of GPS Test.

>Please take a look and let me know if that meets your needs.

Will do, after I upgrade to a version with the DMM format in it.

barbeau, 2018-12-27:
@mirabilos Yes, the "Send location" is the feature I'm referring to (it uses the standard Android "share" icon). If you don't see "copy to clipboard" you can just choose email and the link will pop up in your email app. It's independent of the displayed format so you should be able to try it in the current version of GPSTest. Enter the link in your browser and it should take you to the GeoHack site. Although skimming now I don't see DMM format on GeoHack site...

mirabilos, 2018-12-28:
Sean Barbeau dixit:

>@mirabilos Yes, the "Send location" is the feature I'm referring to
>@(it uses the standard Android "share" icon). If you don't see "copy
>@to clipboard" you can just choose email and the link will pop up in

I don’t have eMail set up on Android so this doesn’t work.

>@your email app. It's independent of the displayed format so you

The idea was to have both a link AND plain coordinates in it.
I normally just delete the link part, but it’s useful to
quickly send to someone.

goldfndr, 2019-01-25:
https://github.com/wiglenet/m8b has UTM (and MGRS) conversion [with the MIT license](https://law.stackexchange.com/questions/6081/can-i-bundle-mit-licensed-components-in-a-apache-2-0-licensed-project). Under geodesy.

commit 6d89f7a3 (2019-09-20): Fix #242 - Support Degrees Decimal Minutes coordinate format (#344)
closed 2019-09-20 by barbeau