barbeau/gpstest #244: API 7.0+ On/Off MSL Altitude Not Displayed

danieldjewell, 2018-12-29:
**Summary:** 

Running Android 8.0, Samsung S9+ (Qualcomm). When Android 7+ APIs are enabled in settings no Altitude (MSL) is displayed. When disabled, both altitudes display. 

**Steps to reproduce:** 

Enable/disable APIs in settings

**Expected behavior:** 

Show both regardless (since the MSL comes from the NEMA messages there appears to be an issue receiving/processing NEMA messages from the new API?)

**Observed behavior:** 

No MSL altitude when Android GNSS 7.0+ API enabled

**Device and Android version:** 

Stock Samsung S9+ (Qualcomm), Android 8.0 

**Screenshots:** 

Let me know if you'd really like one -- but it seems self explanatory :). 

** Notes / Thoughts

I did a quick review of the code to try to track this down -- my best guess is that using the new APIs, somehow the NEMA messages aren't being processed and/or received. If it's the API that's borked, one (not very pretty) solution might be to use *both* APIs to get the data (not sure if it will even let you open both objects simultaneously). 

**Side note:** I'm not a GNSS expert by any means - however, I'm familiar with the general difference between the ellipsoid height (e.g. the WGS84 GPS altitude) and the "compensated" (?) altitude. It *appears* that the Altitude reported by the GNSS APIs is not the ellipsoid altitude but is the altitude corrected for the difference in the geoid? And that the NEMA message altitude (used by the "MSL" display) is the raw WGS84 altitude? Or perhaps I have that backwards. If possible, some clarification might be helpful. (I have to go back and read the Android API docs now to see whether the docs specify one or the other. :))

barbeau, 2019-01-28:
This might be related to https://github.com/barbeau/gpstest/issues/253, which I just fixed.  I'm going to close this.  Please let me know what you see after the next beta release.

closed 2019-01-28 by barbeau

danieldjewell, 2019-02-20:
Looks good to me! Showing up correctly now - including on Android 9/Pie.

barbeau, 2019-02-20:
Great, thanks for letting me know!