barbeau/gpstest #246: French translation

bugeaud, 2018-12-30:
I've made a first French translation of GPSTest as a small contrib

See the for in https://github.com/bugeaud/gpstest

Res files are in 
https://github.com/bugeaud/gpstest/tree/master/GPSTest/src/main/res/values-fr

Anybody to perform a review & test ?

barbeau, 2018-12-31:
Thanks @bugeaud for working on this!  Could you go ahead and put this in a pull request?  I'll do a quick review there and then I can include it in the next beta release.

goldfndr, 2019-01-06:
Should line 286 have a \u202F ?

bugeaud, 2019-01-09:
@barbeau I'll make the pull request, I will just check before with @goldfndr if there is any issue. Please note that I used advanced unicode codepoint to ensure correct typography in the French version (thus the u202F). But as I did not test inside the application, you will need to check that the visual rendering by the stack is consistent with the expectation.

Hello @goldfndr, line 286 of which file/version. Current -fr version for strings.xml hold a max of 274 lines.

goldfndr, 2019-01-12:
My apologies. Line 186 (not 286) has "!" without space.

barbeau, 2019-07-12:
We ran into a license issue accepting this particular text in PR https://github.com/barbeau/gpstest/pull/250, so I'm closing this issue out.  If someone else wants to contribute a French translation I'd love to accept it - see https://github.com/barbeau/gpstest/blob/master/TRANSLATIONS.md for instructions.

closed 2019-07-12 by barbeau