barbeau/gpstest #247: Missing flag

Giolly90, 2019-01-09:
**Summary:** 

Flag for satellite 331 is missing, please also change the flag for all Galileo satellites with a standard EU flag, not the Galielo one it will be consistent with the rest (plus flags are too small and not easy to understand what is shown)

**Expected behavior:** 

Show EU flag

**Device and Android version:** 

OnePlus5 OxygenOS Android Pie 9.0.0

barbeau, 2019-01-09:
@Giolly90 Interesting! I can't find a reference to 331 anywhere online. 

Is this a stock install of Oxygen OS on your OnePlus5? And what country are you in?

A missing GNSS flag on Android 9 stock install means that the GNSS isn't recognized by the Android OS. Top contender would be IRNSS as it doesn't yet have an Android system constant.

Anyone else have a guess for what this is?

Giolly90, 2019-01-10:
Yes, I'm on stock OxygenOS and the country is Italy.
In my opinion satellite 331 is Galileo FOC FM18, isn't it?

barbeau, 2019-01-15:
@Giolly90 I'm not sure - do you always see this satellite without a flag, or does it only occasionally appear?

Either way, if it is Galileo and you're on stock Android OS, this is a bug in your device and you should report it to OnePlus.  Their firmware isn't reporting the GNSS constellation correctly to Android for that satellite.

closed 2019-01-15 by barbeau

Giolly90, 2019-01-15:
I saw 331 only once, I'll give you update if I see it again.

What about the EU flag request? Can you change with this?

barbeau, 2019-01-15:
@Giolly90 I'm considering changing the Galileo satellites flag to the EU flag...But no decision yet.

Giolly90, 2019-01-16:
The Galileo one is a symbol, like these two, the flag is another story
[image]
[image]

Giolly90, 2019-01-21:
331 spotted again!

Giolly90, 2019-02-09:
Keep spotting 331