barbeau/gpstest #257: Support easy copy/paste of coordinates from Status screen

barbeau, 2019-02-07:
**Summary:** 

This has come up in a few discussions lately, including:
https://github.com/barbeau/gpstest/issues/242#issuecomment-450158918

...and feedback from another user:

~~~
I would have found it very useful if I would have been able to:

Copy text from the "STATUS MENU," or any other I would say. 

I actually got frustrated with not being able to copy the coordinates. Was using in conjunction with "GeoPaparazzi"JFYI...... 

Android Oreo 8 (Samsung Galaxy S9) 
~~~

We currently have a "Share" icon that pops up in the top action bar after you get a fix, which will give you a lot of options to share a URL formatted for GeoHack, which is a meta-formatter for a lot of different services (including OSM and Google Maps):

https://tools.wmflabs.org/geohack/geohack.php?params=29.72629576;-81.9470374

One of the share options is also "Copy to clipboard", which will copy the URL to the clipboard.

But it seems users also prefer having a quick copy/paste solution from within the app

**Steps to reproduce:** 

Try to copy/paste coordinates after a fix from within the Status screen

**Expected behavior:** 

Allow me to copy/paste coordinates

**Observed behavior:** 

You can't copy/paste coordinates

**Device and Android version:** 

N/A

GeoZac, 2019-03-24:
I was playing with Android Studio and maybe I could suggest a method
I changed the layout a bit, i.e, moved the latitude and longitude to same TableRow element

[image]

the coordinates can be copied easily by now adding a onClickListener to this row and formatting a string with coordinates and pushing it to ClipboardManager service  as ClipData 

[image]

GeoZac, 2019-03-24:
You could change the layout a bit, i.e, moved the latitude and longitude to same TableRow element.

The coordinates can be copied easily by now adding a onClickListener to this TableRow and formatting a string with coordinates and pushing it to ```ClipboardManager```  service  as ```ClipData```

barbeau, 2019-10-08:
With https://github.com/barbeau/gpstest/commit/99e0419214ec03863b77fdf996a3651324d5f824, now you can simply tap on the top card and it will automatically copy your location to the clipboard.

commit 99e04192 (2019-10-08): Fix #257 - Copy location to clipboard on tap of Status location CardView
closed 2019-10-08 by barbeau