barbeau/gpstest #258: New Galileo satellites issue

Giolly90, 2019-02-12:
**Summary:** 
As I previously mentioned with regard to satellite n.331 is now confirmed: I observed the same behaviour for another satellite of the Galileo costellation which has gone available yesterday. It is satellite n.336 aka Galileo-FOC FM19 "Tara". (PRN E36 is the code where to look the number displayed in GPSTest App, where Galileo satellites get the number 3+PRN ==> 336 in this case).
Also here flag is missing and I am sure the same behaviour will appear for the other newely activated satellites: 313, 315 and 333.

[image]

[image]

Summarize your issue in one sentence (what goes wrong, what did you expect to happen)

Please if possible add EU flag to this new series of satellites

Let me know also about your decision whether changing the Gaileo symbol with the EU flag, here is another example which I want to bring in support of my request.

[image]

Thank you,

Giolly90, 2019-02-12:
Satellite 331 is Galileo-FOC FM18 "Irina" 3-31

barbeau, 2019-02-14:
@Giolly90 Apparently this is a fairly widespread issue with Galileo and some receivers not properly registering the new SVs:
https://www.gsc-europa.eu/news/informative-note-about-galileo-svids-tracking

Please reach out to your device manufacturer and ask them to fix this.  Thanks!

Re: EU flag - I plan to make the change.

barbeau, 2019-02-23:
@Giolly90 I forgot to ask one question - can you please make sure "Use GNSS APIs" setting in GPSTest is checked?

Looks like you might see this on some devices if you uncheck this setting:
https://groups.google.com/d/msg/gpstest_android/pjZV80hlbwI/Pj1GPzg9BQAJ

Giolly90, 2019-02-24:
@barbeau thanks for suggestion and sorry for late reply, but I was testing a bit. I checked the mentioned option and now the phone is correctly recognizing all satellites! (At least I have evidence of 331 and 333, but I think also 336 will be ok).

BTW I noticed that enabling GNSS APIs the numbering is different and the first 3 is omitted.

Anyway I think this finally solved the issue! (I also opened a fix request on OnePlus page, let's see what happens).

Let me know of the changing EU flag request ;)

barbeau, 2019-02-28:
@Giolly90 Good to hear!  Sorry, I assumed you had the default settings for the app, and I didn't realize that this would show up in this way from the legacy API until Lodro posted in the Google Group.  I do have control over the way the Galileo flag matching happens on the legacy API, so I'm going to re-open this issue so I can fix that.  You can close the OnePlus issue and just say that you observed this in an app when a setting was changed to use the legacy GpsStatus API, instead of the newer GnssStatus API.

Re: EU flag - see https://github.com/barbeau/gpstest/commit/3dd3def05027e8a2d91b755f6c1d663183bbcdc0 😉

Giolly90, 2019-03-01:
Great! Thanks for your commitment

barbeau, 2019-03-18:
Including https://github.com/barbeau/gpstest/issues/264 and https://www.gsc-europa.eu/news/informative-note-about-galileo-svids-tracking, here's the total list that needs to be added for Galileo:
* 331
* 332
* 333
* 334
* 335
* 336

commit 2d36a400 (2019-03-19): Fix #258 - Add new Galileo satellites for legacy GpsStatus API
closed 2019-03-19 by barbeau