barbeau/gpstest #267: There is no indication of SBAS data usage in the fix (and not in FAQ)

gdt, 2019-03-12:
**Summary:** 

I want to know if my device is computing a fix based on just GNSS satellites, or if it is applying corrections based on SBAS (or other).

**Steps to reproduce:** 

Run gpstest (up-to-date from f-droid).   Observe that on first fix, error estimates might be as high as 15m ish, and then rapidly settle to 10m.  After quite a while, consistent with receiving corrections from SBAS, I often see estimates down to 3m.    So fairly obvious the device is using SBAS corrections in that mode, but there is no way to tell other than inferring from a range error obtained by backing out HDOP from accuracy.

Use a Garmin, such as an Etrex 30.  Note that satellites being tracked are shown with hollow or solid bars, which either means used in fix or have ephemeris, and a D is shown if the satellite has differential corrections.  This is of course not necessary to understand the issue; it is clearly conceptually useful to know if the solution includes differential corrections.

**Expected behavior:** 

I expect an indication per satellite if there are corrections available and/or being used.   I realize that with limited APIs that may be difficult.

Or, if this is not possible, I would like it to be addressed in the FAQ, which says that SBAS is "supported".

**Device and Android version:** 

Pixel XL, Lineage 15.1.    Also seen on Nexus 5, Lineage 14.1.

goldfndr, 2019-03-15:
If an SBAS satellite is in use, it'll have a "U" flag, just like the GNSS satellites. (if your ROM/chipset doesn't have a bug that doesn't properly implement the flag)

See #108 for a couple of screenshots: the first one has the U flag for an in-use SBAS satellite (46), the second one has 4 meter accuracy without using SBAS. I've frequently seen three (46, 48 51) with AE but only ever one with U.

gdt, 2019-03-16:
@goldfndr  I think you are referring to doing ranging from the SBAS transmitter (which is also interesting).    I am asking about knowing if the differential corrections transmitted via SBAS are being applied to the pseudoranges observed to other satellites.

barbeau, 2019-03-17:
@gdt Yes, this is a good point, and something that's not 100% clear in the current Android spec.  Here's the boolean value for "used" that's passed from the chipset to the Android OS to GPSTest, which is shown as a "U" letter in the "Flags" column on the GPSTest Status screen:
https://developer.android.com/reference/android/location/GnssStatus.html#usedInFix(int)

It says:

>Reports whether the satellite at the specified index was used in the calculation of the most recent position fix.

My assumption is that this means the L1 signal (or other CF) being broadcast from the SBAS satellite is being used in ranging and used in computing the position solution.  However, I'm not 100% sure that's how it's being implemented by OEMs.  I can add something to this effect in the FAQ.

The above is the only data provided by the Android APIs for a SBAS satellite for if it's being "used" - we currently do not have any explicit information from the device or Android APIs for whether correction information from a particular SBAS satellite is being used, and for which satellite it's being applied.

barbeau, 2019-03-17:
Also, note that, in theory, the correction information from SBAS satellites could also be acquired in the A-GPS assistance information over SUPL or other channels.

gdt, 2019-03-17:
Thanks.  I agree that the API docs imply that it's about the use of pseudoranges from that sat in the navigation solution.  Fixing the FAQ seems sufficient for "explain what used means" in this case - that the spec says something, and it's hard to tell what the ROM is doing.

gdt, 2019-03-17:
Off topic, but if you can explain the use of IP vs SUPL for A-GPS, and also what information from the phone is transmitted when A-GPS is requested, that would be really useful.   I would hope that there is no transmission of the current best guess of coordinates to Google, but I have no confidence.

closed 2019-03-19 by barbeau

barbeau, 2019-03-26:
@gdt Re: SUPL - I'd like more info about this myself.  It's been a while since I've done a deep dive of what's transmitted over the assistance channels.  I posted what I knew at the time in this Google Group post - https://groups.google.com/d/msg/gpstest_android/wOgMdQ0N_vU/lU0Qnkq48zIJ.

re: privacy, this analysis claims identifying info like IMEI and location info like serving cell is sent to Google (or at least was at the time):
https://blog.wirelessmoves.com/2014/08/supl-reveals-my-identity-and-location-to-google.html

commit 88c3dd89 (2019-03-19): Fix #267 - Clarify what "Used" means for SBAS satellites in FAQ