barbeau/gpstest #268: Error in AccumulatedDeltaRange Xiaomi Mi 9

Atlanta88, 2019-03-14:
Hi,
For my study, I acquired the latest Xiomi Mi 9. However, when I wanted to process the data acquired under GNSSlogger, I realized that AccumulatedDeltaRange are missing and I don't know why. So I can't do any post-processing. The Android version of the phone is 10.2

Thank you in advance for your answers

[image]

barbeau, 2019-03-17:
@Atlanta88 This issue tracker is for the GPSTest app, not Google's GnssLogger app (see https://github.com/google/gps-measurement-tools/tree/master/GNSSLogger for that app).

That being said, the table at https://developer.android.com/guide/topics/sensors/gnss doesn't list the Xiaomi Mi 9 as supporting accumulated delta range, although the Mi 8 is listed as supporting it.  The Mi 8 used a Broadcomm chip instead of the Snapdragon 845 chipset for GNSS, so it's possible that in the transition to using the SD 855 for GNSS in the Mi 9 that it lost support for accumulated delta range, although I hope this isn't the case.

Have you tried enabling developer mode in the Android OS and checking "Use Full GNSS Raw Measurements" option?

closed 2019-03-17 by barbeau

ionkarolos, 2019-05-02:
Hello did you solve the problem with the accumulated delta range on Xiaomi Mi 9. I would like to buy this phone in order to implement a PPP wizard on this device so I need access to carrier phase measurements. I have already the Xiaomi Mi 8 and doesn’t have any problems regarding with accumulated delta range. Thanking you in advance.

Atlanta88, 2019-05-06:
Hey,
For the moment, no feedback has allowed me to operate the phase measurement. The last update of the phone did not a priori change anything. Looking at the python code, it seems that 16 in AccumulatedDeltaRange meaning an android error on the measurement of time. 
I haven't found any solution yet, I make a post on the issue tracker of Goggle, but no answer for the moment...

barbeau, 2019-05-06:
@Atlanta88 Could you add a link to your Google issue here?

Atlanta88, 2019-05-07:
@barbeau Yes of course : https://issuetracker.google.com/issues/128623751

PedroVenancio, 2019-11-09:
Tested in MIUI 11 and AccumulatedDeltaRange is still 0.

So, no hope with Mi9.

alpterlab, 2020-05-02:
Hello ! It must be a reason why the Mi9 wont allow ADR (carrier phase) and his little brother Mi8 it does...  In the market report of the [Mi9 does state ](https://www.mi.com/uk/mi9/specs)all the goods about dual freq GPS....
Therefore there must be either a technical or commercial reason....
Any idea @benzfish  ?
Something it seems to be updated regarind GPS data sensor... maybe this could be also  implemented.... 
#336

Shuo-Tang, 2021-07-06:
So how can we use accumulated delta range instead of carrier phase? The API explains ADR as another form of carrier phase measurement, but I find in the Rinex file, the carrier phase measurement usually has a big value and is close to `code measurement/wavelength`. The ADR has a positive/negative small value with the same trend with code measurement. How can we utilize that on earth?

barbeau, 2021-07-06:
@Shuo-Tang That's a really good question, and worth asking on the Google Group too:
https://groups.google.com/forum/#!forum/gpstest_android

I've never actually done this myself, but I believe the ADR value is just an offset for the computed pseudorange value?

Take a look at the source code that the Google GNSS measurement tool uses to process ADR:
https://github.com/google/gps-measurement-tools/blob/063e2275e29de1f5476f2a9a9e0cb900a37d0b68/opensource/ProcessAdr.m

Here's the Android ADR API definition:
https://developer.android.com/reference/android/location/GnssMeasurement#getAccumulatedDeltaRangeMeters()

The EU GSA raw measurements document really doesn't go into ADR in detail:
https://www.euspa.europa.eu/system/files/reports/gnss_raw_measurement_web_0.pdf

I'd certainly welcome input from anyone who has computed carrier phase measurements from ADR themselves.