barbeau/gpstest #269: Give indication of whether differential corrections are in use

gdt, 2019-03-17:
In #267, there are two possible meanings for "use of SBAS".  One, which #267 was clarified to mean, is about whether the U flag for an SBAS satellite means that pseudoranges from that satellite are used in the navigation solution (our current belief) or if it means something else.

There is an entirely separate issue, which is whether and how the user can tell if differential corrections for a particular satellite are being used in the navigation solution, analogous to the D indication in Garmin receivers.  This issue is about that indication or lack thereof, being either implemented or clarified as known to be impossible in the FAQ.

As @barbeau pointed out,The "D" indication has a further subtlety about differential corrections being available via SBAS and/or assist data.  (My own experience (pixel xl) is that they are available over the SBAS channel after it has been received solidly for some significant number of minutes, and not available via assist.)

barbeau, 2019-03-18:
Thanks @gdt!  Yes, as mentioned in https://github.com/barbeau/gpstest/issues/267#issuecomment-473719378, knowing whether differential corrections are being applied is not currently supported in the Android APIs.  I'll leave this issue open so if in the future Android support is added we can add support to GPSTest.

gdt, 2020-11-22:
I was just reading the NMEA logs (yay! for the log recording facility), and noticed that in GPGGA there is a field with value 1/2/4/5 for autonomous/differential/rtk-fix/rtk-float.    So while that might not be reliable, showing one of "AUTO", "DIFF", "FIX", "FLOAT" would be great, and even better to figure out a better word for autonomous.   I am seeing 2.

barbeau, 2020-11-23:
@gdt Interesting! Which device is this? Can you share some of those logs? And which NMEA docs are you using for the RTK values?

gdt, 2020-11-23:
See https://www.gpsworld.com/what-exactly-is-gps-nmea-data/ that mentions the RTK values.  I don't have any actual NMEA specs.   I will try to to figure this out and make a more coherent report.

On a Pixel XL, using GPSTest's log recording feature, I am always seeing "2" from GPSTest on my phone even on startup.   I can try it again after not having used GPS for a long time.    I know with gpsd I have seen "3D" vs "3D differential" displayed, so the notion that differential is visible in NMEA did not seem surprising.   Of course, it may be that Android does something odd and always reports SBAS when it isn't true.

gdt, 2020-11-23:
I amend my suggestion to use "SINGLE" for codepoint 1 instead of "AUTONOMOUS".   It might be that just showing this value from GGA as tokens or even numbers in a field "NMEA Fix Mode:" would lead to lots of people noticing things and figuring out more, if you're up for a change  based on a blog post about a spec (which feels 99% reliable to me).

gdt, 2021-06-03:
Hey bot: Despite no activity, this is still valid.

gdt, 2022-04-20:
Hey bot: Despite no activity, this is still valid.

I have modified Vespucci to adjust the accuracy value based on FLOAT/FIX and can confirm that these values are real in the NMEA stream from an F9P.   Particularly with external bluetooth/TCP support, it makes sense for GPSTest to show this field.

barbeau, 2023-12-19:
Not stale

gdt, 2025-01-24:
Hey stale see #645.

gdt, 2026-03-01:
Hey @stale, I commented and you closed this anyway!