barbeau/gpstest #271: SBAS for Australasia

kimollivier, 2019-03-31:
**Summary:** 

Australia and New Zealand now have SBAS on PRN 122

**Steps to reproduce:** 

This does not show as available

**Expected behavior:** 

List 122 in NZ as available

**Observed behavior:** 

SBAS not available

**Device and Android version:** 

Samsung Note 8

**Screenshots:** 

Can be created by pressing the Volume Down and Power Button at the same time on Android 4.0 and higher.

barbeau, 2019-03-31:
Thanks! Can you add a screenshot of this on the Note 8?

barbeau, 2019-04-12:
@kimollivier From a closer look at your screenshot, it looks like your device hardware doesn't support the Australia SBAS, otherwise it would show the record in the SBAS card but the flag would be missing.  I've been holding off adding support for specific satellites like this to GPSTest until there is evidence of hardware support.  So, I'll keep this issue open until we can find a device that actually supports it.

barbeau, 2020-03-04:
News on this front:
https://www.minister.industry.gov.au/ministers/pitt/media-releases/australia-nz-boost-satellite-positioning-technology

Looks like the new SBAS will be called the "Southern Positioning Augmentation Network", or SPAN.

LukePrior, 2022-09-27:
Hi, I think this should be reopened as open services on southpan have recently been enabled.

https://communication.ga.gov.au/link/id/zzzz633276f5c3808450Pzzzz5eeb356d5558b601/page.html

https://communication.ga.gov.au/link/id/zzzz633276f5c4b2a114Pzzzz5eeb356d5558b601/page.html

closed 2023-04-20 by barbeau