barbeau/gpstest #276: Russian translation

jabozkzksl, 2019-04-16:
i hope it works

[russian.zip](https://github.com/barbeau/gpstest/files/3087282/russian.zip)

barbeau, 2019-04-17:
Thanks @jabozkzks!  If you can comment "Yes I agree" in this thread to the CLA that is here:
https://gist.github.com/barbeau/6c075d0118ad00bdb307442fe43ace30

...I can manually add these files to Transifex, which is where translations will be managed going forward.

barbeau, 2019-05-01:
@jabozksl Friendly ping - do you agree to the CLA above for contributing the Russian translation?

barbeau, 2019-05-01:
@jabozksl Great!  I've uploaded them here:
https://www.transifex.com/sean-barbeau/gpstest-android/language/ru/

From now on, you can use Transifex to do the translation, which should make it much easier.

Please take a look at the Transifex link - I think there are a few strings that were added after you did your initial translation.  If you can add this via Transifex I'll then download and load them into the app for testing.

barbeau, 2019-05-09:
@jabozksl Thanks for working on this! Here's a version to test that includes the Russian strings:
https://www.dropbox.com/s/abry1x44hz7eufh/GPSTest-google-debug.apk?dl=0

Could you please work on shortening some of the strings?  You can see that even on a wide device like the Samsung Galaxy S8+ some of the strings are too long and cause wrapping, overlapping, or pushing values off-screen:

[image]

jabozksl, 2019-05-09:
Hi, I've made some changes based on your screenshot, because the app, which I got from dropbox, doesn't show interface in Russian. But German interface works well as I tested on my devices.

barbeau, 2019-05-10:
@jabozksl What Android system locale (language) is selected on your device?  The Russian translation should be showing up by default within the app if you have the Russian language set system-wide on your device.  

When testing the Russian translation you should either change your Android system locale to "Russian" in your main Android system Settings menu, or use the new GPSTest app Setting for "Language" to change to Russian.

jabozksl, 2019-05-10:
Yep, locale is set to Russian and I can't find any "Language" section in Settings of GPSTest. Here are some screenshots
[image]

jabozksl, 2019-05-10:
And here is the MD5 of apk, which I've downloaded here
b3beb65052a0d9fa2880ed6c32e9e931  GPSTest-google-debug.apk

barbeau, 2019-05-10:
Ok - could you please try uninstalling your current version and install this one:
https://www.dropbox.com/s/z7q9jcmdqp6pp87/GPSTest-google-debug.apk?dl=0

You should see the Language setting in GPSTest under the "Use Dark Theme" option in Settings - if you don't please let me know.

jabozksl, 2019-05-10:
Got it, but still no Russian or even German now
[image]

barbeau, 2019-05-10:
Alright - let me take a closer look at what's going on

barbeau, 2019-05-14:
@jabozksl Here's a screenshot of the latest Russian translation on a Samsung Galaxy S8+:

[image]

Closer, but we still need to trim off characters to avoid the top right column from extending off the page and the column headers in the 2nd card to get the headers to align right with the columns.

Please try this APK, which should include the Russian language translation and option to change locales:
https://www.dropbox.com/s/vmug0ixkavm375u/GPSTest-Russian.apk?dl=0

barbeau, 2019-05-14:
Also, on the Accuracy screen, would it be possible to reduce the summary text to one line so the bottom of the button isn't cut off?  It's currently cut off because the summary text overflows to two lines:

[image]

jabozksl, 2019-05-14:
Done. It must look much better now.

barbeau, 2019-05-15:
@jabozksl Thanks, that's much better!

[image]

I'm going to close this issue and will work on pushing out a new beta with the translation.

closed 2019-05-15 by barbeau